analogdevicesinc/msdk #259: docs: Mac M1 homebrew install of openocd inside rosetta

thinking-binaries, 2022-11-21:
Just a note, that in this file: [readme](https://github.com/Analog-Devices-MSDK/msdk#readme) the M1 install instructions suggest installing homebrew in a rosetta window. This does work, but there is a small wrinkle you have to be aware of if you already have the arm64 version of HomeBrew installed, to do with paths.

If your instructions are followed to the letter, they do work, but there are traps for the unweary.

The x86_64 version of homebrew installs into /usr/local, and the arm64 version of Homebrew installs into /opt/homebrew/ and both will try to change paths in your .bash_profile. So you can still end up running the wrong version of brew.

To confirm which version is running, you can type `which brew`. To get your install to work correctly when I already had two HomeBrew's installed, it does work if you:

`cd /usr/local/Homebrew/bin` and run that specific i386 brew as follows:
`./brew install libusb-compat libftdi hidapi libusb`

Then starting the EclipseIde with the ./run_eclipse and building/debugging does seem to correctly find openocd, and openocd does find the i386 versions of the libhidapi.dynlib correctly and debugging works. (If you get this wrong, the error states that libhidapi.dynlib could not be opened by openocd).

It might be worth clarifying this in the setup instructions, as I got in a bit of a tangle with two HomeBrew installs (as I use the arm64 Homebrew for most other things), and clearly the i386 (x86_64) openocd executable won't load and run the arm64 version of the libhidusb.

Also, it would be worth noting the reason for this in the install instructions - openocd runs as a i386 (x86_64) executable under rosetta on the M1, and anything running inside rosetta must by definition run x86_64 dynlib files. You can't install x86_64 libs from the arm64 HomeBrew, which is why we have to install a second version of HomeBrew from inside rosetta.

HomeBrew arch x86_64 and arm64 usefully install the binaries and Cellar and other files into completely different parts of the filing system, so they can be made to run concurrently.

As a final request, could you update TUT6426 (Getting Started) to also reflect this install workflow for Mac M1 users?

https://pdfserv.maximintegrated.com/en/an/TUT6245.pdf

Jake-Carter, 2022-11-21:
Hi @thinking-binaries, thank you for your request.  We are limited to Intel silicon locally, so these insights are very valuable.

I've opened a pull request to update our readme with some better instructions for M1 platforms.  This section is a bit of a balancing act between brevity and covering all the possible configurations on a user's machine, since we'd also like to lean on the excellent Homebrew documentation.

In any case - the new section following your suggestions is a definite improvement.  Let me know if I have misrepresented anything.

I will start the process for updating the next rev of TUT6426 with this info as well.

thinking-binaries, 2022-11-22:
Thanks @Jake-Carter I have approve-reviewed your instructions here: https://github.com/Analog-Devices-MSDK/msdk/pull/262#pullrequestreview-1190696986 thanks very much for doing this.

closed 2022-11-28 by Jacob-Scheiffler

Jake-Carter, 2022-11-28:
Thanks @thinking-binaries, we just merged into main.