analogdevicesinc/msdk #260: status of MaxIDE on your website

thinking-binaries, 2022-11-21:
I am currently evaluating your MAX32665 platform and Eclipse IDE. I was hoping to understand whether this product is still actively supported, as the IDE bundle downloaded from your website here: https://www.maximintegrated.com/en/design/software-description.html/swpart=SFW0018610A is at version 1.0.0 and dated 2019-09 (which is 3 years out of date compared to what is in this github repo).

The documentation folder has limited information in it.

There is no getting started guide inside the documentation folder. I found TUT6245 on your website: https://www.maximintegrated.com/en/design/technical-documents/userguides-and-manuals/6/6245.html but that is dated 2019 and has some significant errors in it:

1. Importing projects (as opposed to using FILE/NEW/MAXIM) does not correctly set BOARD and other identifiers, and defaults to the wrong EvKit1. The second method of loading an example does work.
2. When the BLE_DATS project is loaded, it does build and run, but when we return to the C++ design perspective, all of the include files are marked as not found in the editor, and a number of identifiers are undefined (how did it even build?)
3. The getting started guide suggests two methods of fixing this (why is it not already correct at point of install?), the first suggests moving the CDT provider to entry 2 (that doesn't work it says "The provider is locked") and the second method suggests adding a range of paths. These paths refer to MaximSDK/Firmware/... and there is no Firmware folder in that tree. All of these paths are wrong in other ways, including one that has a missing backslash.
4. There was no help in the getting started guide for installing on a Mac, and no help for the M1. Running the debugger openocd from the M1 crashes with an unsatisfied dynlib for libhidapi and much experimentation with HomeBrew led to a dead end. I later discovered this github repo that has correct Mac M1 instructions - why are these instructions not in the getting started guide?
5. Once we got all that working, the wrong paths is clearly causing a significant problem in the BLE_dats example as the debugger single steps through greyed out code guarded by macro definitions that are clearly significantly out of sync between what the editor sees and what the debugger sees.

A quick way for you to resolve and answer these points, might be to create an updated and corrected getting started guide, make sure the include paths issue is fixed out of the box, and release a new bundle on your website.

I was also surprised when I discovered this github repo, that it was not linked in any of the documentation on your website - why not? If I had known of it's existence earlier on I could have self-helped a lot more.

There is also a confusing endless loop of links on your website, documents that vaguely refer to where to go to download the IDE, and it took many circles round this loop before I landed on the correct page with the download for the IDE for Mac, Windows, and Linux.

My question is twofold - are you serious about supporting this product, and when will a new fixed bundle be available for download from your website with the above issues resolved? I mean that in the nicest way possible really, because it looks like a nice product, but the onboarding experience has been one of the most frustrating experiences I have ever had engaging with a new supplier in my 35 years of being a firmware engineer.

thinking-binaries, 2022-11-22:
I appreciate there is a lot to absorb here. A simple way to move forward on this, might be to turn the above checkpoints into a work package for the packaged eclipse-ide and docs folder. It would also be helpful to have a bit of visibility of any planned update dates to the self installing package on your website, just so we can align any future work (after our evaluation we are currently doing) to whatever the latest tested/released installer is.

As a signpost, you might also find it useful to perhaps write a short kbase note tagged against all MAX32 platforms that point to the bundled 1.0.0 2019-9 Eclipse installer, pointing people at this github repo - as the majority of developers if they know about this repo will find it valuable and mostly be able to self-help (especially if evaluating the maturity of your platform... once evaluation is complete, it is often the case that a real follow on project can budget some extra time to get the tools installed and configured, if the evaluation highlighted that extra work is required to get going for real).

Jake-Carter, 2022-11-22:
Hi @thinking-binaries, thank you for your feedback.  Yesterday we pushed a public release that should resolve most of the Eclipse issues you've pointed out above.

See the [v2022_11_Release](https://github.com/Analog-Devices-MSDK/msdk/tree/v2022_11_Release) tag, which matches what is currently available from the automatic installer.  You can retrieve the update via the MaintenanceTool program in the root directory or your MaximSDK installation, or via the Github repo.  Our release strategy going forward will be to maintain a release branch that is tagged every time we push to the MaximSDK installer.

Please purge the old projects from your Eclipse workspace and recreate them with the new project wizard, or re-import the latest copies from the SDK.

The issues should be resolved in the new projects, so I'll respond to the documentation-related issues you've pointed out.

---

> I was hoping to understand whether this product is still actively supported, as the IDE bundle downloaded from your website here: [...]

This is very much still actively supported, and the version number listed here is really just an artifact of our legacy software distribution system.  v1.0.0 indicates the version number for the _installer itself_. which has not changed in that time.   However the _components_ that the installer retrieves are frequently updated and have their own version numbers you can track with the MaintenanceTool.

> There is no getting started guide inside the documentation folder. I found TUT6245 on your website: https://www.maximintegrated.com/en/design/technical-documents/userguides-and-manuals/6/6245.html but that is dated 2019 and has some significant errors in it:

I had not realized how outdated the documentation on our website had become.  Many of the information is only relevant to the legacy SDK that preceded this one.  I will see if we can mark some of these documents as deprecated or create new updated revisions.  In the meantime, I will increase the priority of the updated user guide.  I have already completed most of the content and should be able to get this up soon.  The updated Eclipse section is all that's left.

> There is also a confusing endless loop of links on your website, documents that vaguely refer to where to go to download the IDE, and it took many circles round this loop before I landed on the correct page with the download for the IDE for Mac, Windows, and Linux.

Apologies for the confusion.  Several months ago I submitted several knowledge-base articles in an attempt to address this including [KA-16466](https://maximsupport.microsoftcrmportals.com/en-us/knowledgebase/article/KA-16466).  However, it appears that these are not visible from the main page's search bar...  so are essentially useless.  I will follow up with our web team on this.

> I was also surprised when I discovered this github repo, that it was not linked in any of the documentation on your website - why not? If I had known of it's existence earlier on I could have self-helped a lot more.

Completely agreed.  This is another symptom of the larger issue with our outdated website documentation and KBA issues.  I'll follow up on this as well.

Again - thank you for your honest feedback, and apologies again for the confusion.  We clearly have some work to do with the web team to sync our site up with the latest development resources.  While that's happening, I will be working on completing the updated user guide as the highest priority.

Jake-Carter, 2022-11-29:
@thinking-binaries I've uploaded a preliminary universal user guide to my MSDK fork on the [`dev/userguide`](url) branch.

https://github.com/Jake-Carter/msdk/blob/dev/userguide/USERGUIDE.md

This is still very much a WIP, but the primary content for installation and getting started has been updated and consolidated.  I hope this serves as a useful reference for your evaluation while I continue to flesh out the more advanced material.

closed 2022-12-15 by Jake-Carter