analogdevicesinc/msdk #290: i2s dma  hao pop

xinshuwei, 2022-12-09:


Jake-Carter, 2022-12-13:
@xinshuwei It's a difficult issue to debug remotely.

As Jesse mentioned a double-buffered approach with DMA is recommended.

I would also recommend using `f_stream` instead of `f_read`.  From the [f_read documentation](http://elm-chan.org/fsw/ff/doc/read.html): **"If the file needs to be read fast, it should be read in large chunk as possible."**  `f_read` is slow.  See [f_forward](http://elm-chan.org/fsw/ff/doc/forward.html) for documentation and [out_stream](https://github.com/Analog-Devices-MSDK/msdk/blob/bc90c7d6c4a2f54866a67d497cb8d8753013e4e2/Examples/MAX78000/ImgCapture/src/console/console.c#L245) for an example where I implemented the data streaming function to stream to UART.

Additionally, have you verified that the sine lookup table you're sending from is _perfectly continuous_?  I would recommend starting and ending blocks of audio data at the [Zero-Crossing](https://en.wikipedia.org/wiki/Zero_crossing) for a sine wave.

xinshuwei, 2022-12-16:
[Desktop.zip](https://github.com/Analog-Devices-MSDK/msdk/files/10241572/Desktop.zip)
@Jake-Carter  Thanks your reply.
1.I had upload  my file, please check.  
2. "f_stream" instead of f_read   i will try it, thks.

closed 2023-05-03 by Jake-Carter