analogdevicesinc/msdk #291: Low power example is not working in power down mode(FTHR board)

chaitanya-smartrotamac, 2022-12-09:
```
//I changed the following changes as I want to check only power-down mode
#define USE_BUTTON 1
#define USE_ALARM  0

#define DISABLE_GPIO 0 //it configures all GPIOs as input to save power

#define DO_SLEEP     0
#define DO_LPM       0
#define DO_UPM       0
#define DO_STANDBY   0
#define DO_BACKUP    0 // will reset after wakeup`
#define DO_POWERDOWN 1 // will reset after wakeup
```

Now I am using FTHR board, 
Red light on > Clicked the trigger button > Red light off > waiting for 3.0/3.1 to be pressed.
I connected the pin 3.1 to the gnd and found nothing is happening ?
It should be supposed to wake up right? not happening. Is something wrong with my expectation?

aniktash, 2022-12-13:
Unfortunately not, only P3.0 or reset can be used as the wake-up source in Power Down mode. The example and documentation will be updated soon.
You may use the reset instead, which works the same way.

chaitanya-smartrotamac, 2022-12-14:
> Unfortunately not, only P3.0 or reset can be used as the wake-up source in Power Down mode. The example and documentation will be updated soon. You may use the reset instead, which works the same way.

Thank you @aniktash 
We will use P3.0 in our product design.

Next question, this may not be related to the actual question 

While I am testing I found
"#include "led.h" "
and
"#include "pb.h" "

in code path : https://github.com/Analog-Devices-MSDK/msdk/blob/main/Examples/MAX78000/LP/main.c

1) includes paths that are not visible to the Eclipse IDE
2) https://github.com/Analog-Devices-MSDK/msdk/tree/main/Tools don't contain GNU Tools, not required?
 - I added the maxim SDK path also.
[image]

check my includes, I included almost everything. Still there are some red line missing includes.

[image]

Did I miss something?

aniktash, 2022-12-14:
There seems to be some includes missing compared to the original LP example project. You may want to try to add them, or import the original example:
[image]

chaitanya-smartrotamac, 2022-12-15:
yes, you are right @aniktash .
But I thought adding just the /MiscDrivers will add all driver-related files.

Still, I can find some includes are missing, Can you tell me what are they?
[image]
[image]

though they are missing, Still the compiler can see them while building and always has 0 errors.

aniktash, 2022-12-15:
This is just to help the Eclipse IDE indexing. They are not needed to build;  The compiler uses make files to find them.
LED1 and CONSOLE_UART are in include files:
[image]
pb_pin is here: 
C:\MaximSDK\Libraries\Boards\MAX78000\EvKit_V1\Source\board.c
Generally, the indexer should be able to find if it is enabled:
[image]
Sometimes the indexer needs help; You may need to choose to search for unresolved includes, re-resolve or refresh and rebuild:
[image]

chaitanya-smartrotamac, 2022-12-20:
> This is just to help the Eclipse IDE indexing. They are not needed to build; The compiler uses make files to find them. LED1 and CONSOLE_UART are in include files: [image] pb_pin is here: [...]

Wow, That rebuild indexing did the trick. Thank you @aniktash 

This will be my last question in this post chain,

[image]
How to find to which function it was referring?
I tried ctrl+click but it is going nowhere. 

When debugging I can see by stepping in, But is there a way to know without debugging?

aniktash, 2022-12-20:
The peripheral drivers are in this folder: C:\MaximSDK\Libraries\PeriphDrivers\Source
Inside each peripheral folder, you can find the specific driver for MAX78000 tagged as ai85 (it's our internal die name), e.g.
uart_ai85.c. 
For some peripherals, a generic driver is used, or a driver from another die (me17). You can see the list of drivers used for MAX78000 in this make file:
C:\MaximSDK\Libraries\PeriphDrivers\max78000_files.mk

chaitanya-smartrotamac, 2023-01-19:
Thank you @aniktash 
One more last issue. We developed our custom board with MAX78000 and found that JLINK is officially supported. We tried it with no luck.
Is there a example code or project how to setup the Jlink to program max78000?

aniktash, 2023-01-19:
@jessexm could you please support?

chaitanya-smartrotamac, 2023-01-20:
Hi @Jake-Carter 
Thank you for your response.

I will try your method in VS code and sure will let you know any issues with it.

We tried from our side with eclipse and found a working configuration, Which you can use for testing and add top the official document.

Debug configuration
[image]

Debugger settings
[image]

startup
[image]

Common
[image]

That's it, It worked.

I think prerequisites are
- Jlink debugger
- Jlink tools installed

Jake-Carter, 2023-01-20:
Thanks @chaitanya-smartrotamac, this is great.  I'll test it out and get it added to our UG.  Much appreciated!

Saicharan036, 2023-02-15:
Hi @Jake-Carter 

I am trying to setup Visual studio jlink setup and following problems I am facing

1) Cant find open settings json

[image]

2) Build was successful but flash is failing
-----------------------------------------------------------------------------------------------------------------------------------------
This is error : 

 *  Executing task: make -r -j 8 TARGET=MAX78000 BOARD=EvKit_V1 MAXIM_PATH=C:/MaximSDK MAKE=make PROJECT=P3_DSP_Development_P3board_UART_cmd 

Loaded project.mk
arm-none-eabi-size --format=berkeley /c/Users/LENOVO/Max78000_1/P3_DSP_Development_P3board_UART_cmd/build/P3_DSP_Development_P3board_UART_cmd.elf
   text    data     bss     dec     hex filename
  36572    2512    1284   40368    9db0 C:/Users/LENOVO/Max78000_1/P3_DSP_Development_P3board_UART_cmd/build/P3_DSP_Development_P3board_UART_cmd.elf
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: arm-none-eabi-gdb --cd="C:\Users\LENOVO\Max78000_1\P3_DSP_Development_P3board_UART_cmd" --se="build/P3_DSP_Development_P3board_UART_cmd.elf" --symbols=build/P3_DSP_Development_P3board_UART_cmd.elf -x="C:\Users\LENOVO\Max78000_1\P3_DSP_Development_P3board_UART_cmd/.vscode/flash.gdb" --ex="flash_m4 C:/MaximSDK/Tools/OpenOCD jlink.cfg max78000.cfg" --batch 

C:\MaximSDK\Tools\GNUTools\10.3\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the 
index cache directory.
The target architecture is set to "armv7e-m".
Open On-Chip Debugger 0.11.0+dev-g56a818e4c (2022-11-16-05:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Remote communication error.  Target disconnected.: No error.

 *  The terminal process "C:\Windows\System32\cmd.exe /d /c arm-none-eabi-gdb --cd="C:\Users\LENOVO\Max78000_1\P3_DSP_Development_P3board_UART_cmd" --se="build/P3_DSP_Development_P3board_UART_cmd.elf" --symbols=build/P3_DSP_Development_P3board_UART_cmd.elf -x="C:\Users\LENOVO\Max78000_1\P3_DSP_Development_P3board_UART_cmd/.vscode/flash.gdb" --ex="flash_m4 C:/MaximSDK/Tools/OpenOCD jlink.cfg max78000.cfg" --batch" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
 ---------------------------------------------------------------------------------------------------------------------------------------

[image]

3) I changed the Jlink drivers using the process mentioned here
https://github.com/Analog-Devices-MSDK/VSCode-Maxim/wiki/Segger-J-Link-Setup-Guide

how to revert back to normal jlink?

---------------

I added this setting in User settings

[image]

I clicked on erase and I got this
Open On-Chip Debugger 0.11.0+dev-g56a818e4c (2022-11-16-05:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : J-Link V11 compiled Dec 14 2022 09:09:01
Info : Hardware version: 11.00
Info : VTarget = 1.772 V
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x2ba01477
Info : max32xxx.cpu: Cortex-M4 r0p1 processor detected
Info : max32xxx.cpu: target has 6 breakpoints, 4 watchpoints
Info : max32xxx.cpu: external reset detected
Info : starting gdb server for max32xxx.cpu on 3333      
Info : Listening on port 3333 for gdb connections        
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x00002706 msp: 0x20003e38
Error: timed out while waiting for target halted
TARGET: max32xxx.cpu - Not halted

 *  The terminal process "C:\Windows\System32\cmd.exe /d /c openocd -s C:/MaximSDK/Tools/OpenOCD/scripts -f interface/jlink.cfg -f target/max78000.cfg -c "init; reset halt; max32xxx mass_eras
 *  Terminal will be reused by tasks, press any key to close it. 
 
 
 why is it "TARGET: max32xxx.cpu - Not halted" but I have MAX78000

Jake-Carter, 2023-02-16:
Hi @Saicharan036 

> Cant find open settings json

Looks like VS Code changed the name of that option in the latest update, so my screenshot is outdated.  You found the correct option - it's _Open User Settings (JSON)_

> how to revert back to normal jlink?

You can follow steps 1-3 from https://github.com/Analog-Devices-MSDK/VSCode-Maxim/wiki/Segger-J-Link-Setup-Guide#option-2-j-link-with-jlinkgdbserver to revert the adapter back to the normal JLink drivers

> why is it "TARGET: max32xxx.cpu - Not halted" but I have MAX78000

max32xxx is a common CPU configuration across all of our micros, so this is expected.

Everything looks correct on your setup.  There should be a "flash.log" file in your project with some more detailed OpenOCD output on why the flash failed.  Can you attach it?

chaitanya-smartrotamac, 2023-03-10:
Hi @Jake-Carter @aniktash 
Can someone help me quickly with my issues?
I want to interface with a Bluetooth microcontroller to program the MAX78000 using the serial bootloader.
I mentioned everything in this issue here https://github.com/MaximIntegratedAI/secure-loader/issues/3#issue-1618428238

Right now, I want to understand how this works by programming MAX78000 fthr eval kit using another MAX78000 fthr eval kit.

closed 2023-03-31 by Jake-Carter