analogdevicesinc/msdk #315: openocd included in package is not compatible with macOS , issues with Max78000. via cmsis-dap if homebrew openocd is used

pgreendale, 2022-12-25:
Used Platform: MacOS 12.4 Monterey Arm, M1 Macbook Air 
Used Board : FTHR_RevA

Issue: 
If openocd is installed with automatic installer as recommended in MSDK Userguide [link](https://github.com/Analog-Devices-MSDK/msdk/blob/main/Documentation/USERGUIDE.md#download) , it throws 
dyld[79395]: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
...
Reason: tried: '/usr/local/opt/hidapi/lib/libhidapi.0.dylib' (no such file), '/usr/local/lib/libhidapi.0.dylib' (no such file), '/usr/lib/libhidapi.0.dylib' (no such file)

as error, which makes sense to me, there is no /usr/local/lib at vanilla MacOS. 

Installing missing libraries with homebrew doesnt help because cellar paths are different from path searched by openocd executable. Replacing executable openocd bin file with homebrewed openocd 0.11 is responsive and allows register access to MAX78000 with cmsis-dap but throws errors when writing flash. 

My actual workaround is based on [avrdude solution](https://github.com/avrdudes/avrdude/issues/939) to download intel files, extract libusb and libhidapi into /usr/local/lib (create /usr/local/lib dir before) and renaming them as requested by original openocd executable. Original openocd executable provided by MSDK downloader will work flawless then. Seems to me that provided openocd differs from homebrew version, but I wont check for that. If there is a difference, please check. Possible solution add .dylib files to Tools/OpenOCD/lib and fix linker to relative paths within openocd executable?

pgreendale, 2022-12-25:
Update: Close Ticket. Didnt RTFM enough, there is already a way fixing this issue included in the [MSDK user guide](https://github.com/Analog-Devices-MSDK/msdk/blob/main/Documentation/USERGUIDE.md) , section Completing the Installation on MacOS . 
Anyway, the proposed solution may help if you dont want to modify your homebrew path and having multiple terminals / Brew installations (as I like to avoid to have a not too cluttered system). Rosetta needs to be installed and cares about the launch of ix86 compiled libs automatically.

closed 2022-12-25 by pgreendale

Jake-Carter, 2023-01-02:
Thanks @pgreendale.  Yes, our version of OpenOCD is a custom [fork](https://github.com/Analog-Devices-MSDK/openocd) that we've made to add support for our hardware.  I wish we had some M1 silicon to provide native binaries.

It looks like M1-hosted Github runners are in [beta](https://github.com/github/roadmap/issues/528), so we'll take a look at it again.  A dedicated Rosetta Homebrew install is our primary recommendation since it's consistent to set up for most users. 
 Luckily the Homebrew documentation is very good, so we're intentionally leaning on it a bit for solutions like the one above