analogdevicesinc/msdk #321: MAX32690 BLE_fit hard fault

kevin-gillespie, 2023-01-04:
Experiencing hard fault and/or assertions with BLE_fit running on the MAX32690.

https://github.com/Analog-Devices-MSDK/msdk/pull/320

kevin-gillespie, 2023-01-04:
Pretty confident that the issue is software related. Seems to be highly dependent on code placement and execution sequence. 

Issue is still present if we:

- increase the core voltage
- run from the slower ISO
- limit the SRAM usage
- use a different flash bank.

kevin-gillespie, 2023-01-05:
It seems like the way the interrupt handlers are being compiled is significant. We currently have mostly empty interrupt handlers that call a peripheral driver handler. 

In this case there is only one instruction that branches to the peripheral driver handler. In this case we get a hard fault after the interrupt returns.
```
void WUT0_IRQHandler(void)
{
    MXC_WUT_Handler();
10000240:	f025 bca8 	b.w	10025b94 <MXC_WUT_Handler>
```

These functions are compiled differently if there are additional instructions. This works as expected.
```
void WUT0_IRQHandler(void)
{
10000240:	b508      	push	{r3, lr}
    MXC_WUT_Handler();
10000242:	f025 fcab 	bl	10025b9c <MXC_WUT_Handler>
    __asm volatile("NOP");
10000246:	bf00      	nop
}
10000248:	bd08      	pop	{r3, pc}
```

If we add the interrupt attribute, it is compiled a little differently. This works as expected.
```
void __attribute__((interrupt))WUT0_IRQHandler(void)
{
10000240:	4668      	mov	r0, sp
10000242:	f020 0107 	bic.w	r1, r0, #7
10000246:	468d      	mov	sp, r1
10000248:	b501      	push	{r0, lr}
    MXC_WUT_Handler();
1000024a:	f025 fcaf 	bl	10025bac <MXC_WUT_Handler>
}
1000024e:	e8bd 4001 	ldmia.w	sp!, {r0, lr}
10000252:	4685      	mov	sp, r0
10000254:	4770      	bx	lr
```
The WUT_Handler is the same in all cases. We can see that r3 is used in the first instruction. 
```
10025bac <MXC_WUT_Handler>:
    NVIC->ICPR[((uint32_t)(IRQn) >> 5)] =
10025bac:	4b4a      	ldr	r3, [pc, #296]	; (10025cd8 <MXC_WUT_Handler+0x12c>)
10025bae:	f44f 1200 	mov.w	r2, #2097152	; 0x200000

/* ************************************************************************** */
int MXC_WUT_Handler(void)
{
10025bb2:	e92d 41f3 	stmdb	sp!, {r0, r1, r4, r5, r6, r7, r8, lr}
10025bb6:	f8c3 2184 	str.w	r2, [r3, #388]	; 0x184
    int trimComplete;
    mxc_wut_complete_cb_t cbTemp;

    /* Clear the interrupt flags */
    NVIC_ClearPendingIRQ(WUT0_IRQn);
    MXC_WUT_IntClear();
10025bba:	f7ff ffeb 	bl	10025b94 <MXC_WUT_IntClear>
```
```

kevin-gillespie, 2023-01-05:
@EdwinFairchild @lorne-maxim

kevin-gillespie, 2023-01-05:
Maybe what's significant is that it's ensuring that the stack pointer is double word aligned. 

https://github.com/ARM-software/abi-aa/blob/2982a9f3b512a5bfdc9e3fea5d3b298f9165c36b/aapcs32/aapcs32.rst#6212stack-constraints-at-a-public-interface

lorne-maxim, 2023-01-06:
I don't think it is the double word alignment that is the issue here.  If that were the case, then I don't think the NOP workaround would have fixed it.  That code pushes an additional two words (R3 and LR) on the stack which wouldn't change the alignment.

Saving R3 must also not be significant since the __attribute__(interrupt) workaround doesn't save/restore R3 and that code works.

kevin-gillespie, 2023-01-09:
You're right. Still seeing some errors even with the __IRQ. In this case the stack is intact and we get a bus fault error. Looks like we tried to use some stack memory that is over the limit. 

```
(gdb) bt
#0  HardFault_Handler () at /home/kgills/Workspace/msdk/Libraries/Boards/MAX32690/EvKit_V1/Source/board.c:232
#1  <signal handler called>
#2  0x10022228 in memset ()
#3  0x1000a226 in BbBleConnSlaveInit () at /home/kgills/Workspace/msdk/Libraries/Cordio/controller/sources/ble/bb/bb_ble_conn_slave.c:303
#4  0x1001106a in lctrSlvConnResetHandler () at /home/kgills/Workspace/msdk/Libraries/Cordio/controller/sources/ble/lctr/lctr_main_conn_slave.c:134
#5  0x1000e66e in LctrMsgDispatcher (pMsg=<optimized out>) at /home/kgills/Workspace/msdk/Libraries/Cordio/controller/sources/ble/lctr/lctr_main.c:240
#6  0x1001937c in wsfOsDispatcher () at /home/kgills/Workspace/msdk/Libraries/Cordio/wsf/sources/targets/baremetal/wsf_os.c:299
#7  0x100193f6 in WsfOsEnterMainLoop () at /home/kgills/Workspace/msdk/Libraries/Cordio/wsf/sources/targets/baremetal/wsf_os.c:398
#8  0x10000b12 in main () at main.c:270
(gdb) x/x 0xE000ED28
0xe000ed28: 0x00008200
(gdb) print/x *(uint8_t *)0xE000ED29
$7 = 0x82
(gdb) x/x 0xE000ED38
0xe000ed38: 0x201202ec
```

kevin-gillespie, 2023-02-03:
Root cause of the issue was a cache settings. Resolution is in the test program.

closed 2023-02-03 by kevin-gillespie