analogdevicesinc/msdk #345: BLE_Freertos Secured Connection Failure

EdwinFairchild, 2023-01-20:
I am able to establish a bonded encrypted connection once I add another element to wsfBufPoolDesc_t :
```
static wsfBufPoolDesc_t mainPoolDesc[] = { 
                                           { 16, 8 },
                                           { 32, 4 },
                                           { 192, 8 },
                                           { 256, 16 },
                                           { 512, 4 } }; // new element
```

Currently for `BLE_dats` we set the last element in the  struct below to `TRUE`.
```
static const appSecCfg_t datsSecCfg = {
    DM_AUTH_BOND_FLAG | DM_AUTH_SC_FLAG | DM_AUTH_MITM_FLAG, /*! Authentication and bonding flags */
    DM_KEY_DIST_IRK, /*! Initiator key distribution flags */
    DM_KEY_DIST_LTK | DM_KEY_DIST_IRK, /*! Responder key distribution flags */
    FALSE, /*! TRUE if Out-of-band pairing data is present */
    TRUE /*! TRUE to initiate security upon connection */
};
```

However keeping this value as `TRUE` for  `BLE_FreeRTOS` when using a bonded encrypted connection causes it to fail to reconnect to bonded device after a reset. In other words the initial bonding works and the connection is encrypted just fine. But if you reset the devices they naturally  will reestablish the bonded encrypted connection and this is when it fails, again ONLY when we have the slave initiate security to TRUE. 
Backtrace: 
```
170	    while (1) {}
(gdb) bt
#0  HardFault_Decoder (frame=0x2001fed8) at main.c:170
#1  <signal handler called>
#2  0x10024aa4 in memcpy ()
#3  0x1001fe66 in llc_api_tx_data ()
#4  0x1001f78e in PalBbBleTxTifsData ()
#5  0x1000dd12 in lctrSetupForTx (pCtx=pCtx@entry=0x20005c60 <ucHeap+10460>, 
    rxStatus=rxStatus@entry=0 '\000', reqTx=reqTx@entry=1 '\001')
    at /home/eddie/workspace/msdk/Libraries/Cordio/controller/sources/ble/lctr/lctr_isr_conn.c:301
#6  0x1000e738 in lctrSlvConnRxCompletion (pOp=<optimized out>, pRxBuf=0x200052ea <ucHeap+8038> "\001", 
    status=<optimized out>)
    at /home/eddie/workspace/msdk/Libraries/Cordio/controller/sources/ble/lctr/lctr_isr_conn_slave.c:802
#7  0x1000a346 in bbSlvConnRxCompCback (status=<optimized out>, rssi=<optimized out>, crc=<optimized out>, 
    timestamp=2355035, rxPhyOptions=0 '\000')
    at /home/eddie/workspace/msdk/Libraries/Cordio/controller/sources/ble/bb/bb_ble_conn_slave.c:176
#8  0x1001ffa8 in cb_rx_interrupt_handler ()
#9  0x10022a0e in BTLE_RX_RCVD_IRQHandler ()
#10 <signal handler called>
#11 0x100009b8 in vApplicationIdleHook () at main.c:98
#12 0x100231e6 in prvIdleTask (pvParameters=<optimized out>) at Source/tasks.c:3393
#13 0x10023bc0 in ?? () at Source/portable/GCC/ARM_CM4F/port.c:712
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

We can set this to FALSE and still establish a secure connection because the client is it set to TRUE and all seems to work ok, except the speed test does not start, that might be a whole 'nother can of worms.

EricB-ADI, 2023-01-25:
```
(gdb) bt
#0  PalSysAssertTrap ()
    at /home/eric-busch/Workspace/msdk-fork/Libraries/CMSIS/Include/core_cmFunc.h:316
#1  0x100189b0 in WsfAssert (
    pFile=pFile@entry=0x1002d974 "/home/eric-busch/Workspace/msdk-fork/Libraries/Cordio/controller/sources/ble/lctr/lctr_isr_conn.c", line=line@entry=292)
    at /home/eric-busch/Workspace/msdk-fork/Libraries/Cordio/wsf/sources/targets/freertos/wsf_assert.c:63
#2  0x1000dad4 in lctrSetupForTx (pCtx=pCtx@entry=0x200054b8 <ucHeap+7612>, 
    rxStatus=rxStatus@entry=0 '\000', reqTx=reqTx@entry=1 '\001')
    at /home/eric-busch/Workspace/msdk-fork/Libraries/Cordio/controller/sources/ble/lctr/lctr_isr_conn.c:292
#3  0x1000e4e0 in lctrSlvConnRxCompletion (pOp=<optimized out>, 
    pRxBuf=0x20004172 <ucHeap+2678> "\001", status=<optimized out>)
    at /home/eric-busch/Workspace/msdk-fork/Libraries/Cordio/controller/sources/ble/lctr/lctr_isr_conn_slave.c:802
#4  0x1000a0ee in bbSlvConnRxCompCback (status=<optimized out>, rssi=<optimized out>, 
    crc=<optimized out>, timestamp=429302, rxPhyOptions=0 '\000')
    at /home/eric-busch/Workspace/msdk-fork/Libraries/Cordio/controller/sources/ble/bb/bb_ble_conn_slave.c:176
#5  0x1001f450 in cb_rx_interrupt_handler ()
#6  0x100224b0 in BTLE_RX_RCVD_IRQHandler ()
#7  <signal handler called>
#8  0x10000ea6 in LED_Off (idx=idx@entry=1)
```

I cant recreate the issue entirely. I get an assertion as soon as the datc device attempts to form a connection

EricB-ADI, 2023-01-25:
It seems to be the txLen of the header packet is being set to zero and the assertion is triggering. Not sure if this is happening on your side too.

EdwinFairchild, 2023-01-26:
BLE_fit_FreeRTOS has same behavior  when connecting to BLE_datc with security enabled. 
the main culprit is https://github.com/Analog-Devices-MSDK/msdk/blob/4787a0f7020cdfdd918570253847efe5a5a95c54/Libraries/Cordio/controller/sources/ble/lctr/lctr_isr_conn.c#L292

closed 2023-02-24 by EdwinFairchild