analogdevicesinc/msdk #363: can't read AIN0, AIN1, ... analog inputs using HAL

buha, 2023-01-26:
Hi,

I'm trying to get some readings in with MAX32650 of AIN0, AIN1, etc. external channels.

No matter what I provide on AIN0, AIN1, pins of the FTHR board, I always get the raw value 1 as reading.

However, if I perform ADC conversions of internal channels I get meaningful values.

The seqence of calls is:
```
MXC_ADC_Init();
MXC_ADC_StartConversion(ch); // ch is MXC_ADC_CH_0, MXC_ADC_CH_1 etc.
MXC_ADC_GetData(&reading)
```
Here's the readings that I get converted to Volts using `reading * 1.22V / 1024`
```
mwc:ain0 = 0.001191 Volts
mwc:ain0/5 = 0.001191 Volts
mwc:ain1 = 0.001191 Volts
mwc:ain1/5 = 0.001191 Volts
mwc:vcore = 1.122305 Volts
mwc:vdda = 1.218809 Volts
mwc:vddb/4 = 0.795859 Volts
mwc:vddio/4 = 0.459883 Volts
mwc:vddioh/4 = 0.847090 Volts
```

First values are ALWAYS 1 * 1.22 / 1024 = 0x001191.
The rest of the values seem correct.

buha, 2023-01-26:
nevermind, the FTHR board routes AIN0, AIN1 etc. pins to an external I2C ADC, not to the embedded ADC of the microcontroller...

buha, 2023-01-27:
S

buha, 2023-01-27:
@kevin-gillespie any idea ?

buha, 2023-01-27:
Found the reason...

Board_Init() was doing nothing in my firmware and so by comparison with your example, it wasn't initializing the MXC_SYS_PERIPH_CLOCK_GPIO2 (GPIO2.3 and GPIO2.4).

https://github.com/Analog-Devices-MSDK/msdk/blob/main/Libraries/Boards/MAX78000/FTHR_RevA/Source/board.c#L85

Jake-Carter, 2023-02-01:
Thanks @buha, can we close this?

closed 2023-02-01 by Jake-Carter