analogdevicesinc/msdk #375: should WSF_HEAP_SIZE be "closer to the user"

EdwinFairchild, 2023-02-01:
I was having trouble adjusting wsf heap size to make BLE_FreeRTOS use less SRAM
I was trying to adjust here : https://github.com/Analog-Devices-MSDK/msdk/blob/9534dc2e1f69420bfa951f17668a2b40ad607f0f/Libraries/Cordio/wsf/sources/targets/freertos/wsf_heap.c#L38-L46

But turns out that is being originally defined below and overwriting this in the local project.mk has no effect, I think maybe having this less buried would be helpful at least for the BLE_FreeRTOS example : 
https://github.com/Analog-Devices-MSDK/msdk/blob/9534dc2e1f69420bfa951f17668a2b40ad607f0f/Libraries/Cordio/platform/targets/maxim/max32655/build/config.mk#L50

EdwinFairchild, 2023-02-01:
@kevin-gillespie Once I figured out my changes to `wsf_heap.c` were in vain and lowered the original define of `WSF_HEAP_SIZE`   I have been able to get away with :    

 ```
WSF_HEAP_SIZE=0x4000
and for freertos 
#define configTOTAL_HEAP_SIZE ((size_t)(55 * 1024))
```
and still have 5kb of heap to spare , this is just running the  BLE_FreeRTOS example with no other services/chars. That is a bit extreme and cutting it tight but the  point here being we should make it a bit easier for users to find and fiddle with `WSF_HEAP_SIZE`  values for their use cases, preferably through project.mk

Jake-Carter, 2023-02-01:
Looks like everything in `CFG_DEV` gets prepended with `-D` here:  https://github.com/Analog-Devices-MSDK/msdk/blob/9534dc2e1f69420bfa951f17668a2b40ad607f0f/Libraries/Cordio/platform/targets/maxim/build/cordio.mk#L93

We could make `WSF_HEAP_SIZE` overridable from project.mk with:
```Makefile
WSF_HEAP_SIZE ?= 0x10000
CFG_DEV += WSF_HEAP_SIZE=$(WSF_HEAP_SIZE)
```

EdwinFairchild, 2023-02-01:
That looks like it'll be a winner

closed 2023-02-03 by EdwinFairchild