analogdevicesinc/msdk #448: MAX32655:  Hello_World-riscv Fails to Locate common.ld

Metear, 2023-02-27:
This error occurs when I compile the project. How can I solve this problem? Do you know how to solve this error?

d:/maximsdk/tools/gnutools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file common.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
make[1]: Leaving directory '/d/MaximSDK/Examples/MAX32655/Hello_World-riscv'
make[1]: *** [/D/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk:351: /d/MaximSDK/Examples/MAX32655/Hello_World-riscv/build/Hello_World-riscv.elf] Error 1
make: *** [Makefile:52: arm] Error 2
"make -r -j 8" terminated with exit code 2. Build might be incomplete.

Jake-Carter, 2023-02-27:
Thanks @Metear.  I replicated the issue on Windows only.  It turned out to be a bug in our core gcc.mk file on MSYS2.

Here is a zip file with a new gcc.mk file for you to use.  Please copy the file into `Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC` in your SDK and overwrite the old `gcc.mk` file.
* [msdk-448.zip](https://github.com/Analog-Devices-MSDK/msdk/files/10842327/msdk-448.zip)

Here is the fix I added.  We will need to apply this to all our micros, so I will keep this ticket open until that's done.

```diff
diff --git a/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk b/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk
index 180ebbe29..7e854e5a9 100644
--- a/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk
+++ b/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk
@@ -95,10 +95,15 @@ endif
 # auto-generated dependencies. Also if this is Cygwin, file paths for ARM GCC
 # will be converted from /cygdrive/c to C:.
 ################################################################################
-ifneq ($(findstring CYGWIN, ${shell uname -s}), )
+UNAME := ${shell uname -s}
+ifneq ($(findstring CYGWIN, ${UNAME}), )
 CYGWIN=True
 endif

+ifneq ($(findstring MSYS, ${UNAME}), )
+MSYS=True
+endif
+
 # Get the prefix for the tools to use.
 ifeq "$(TOOL_DIR)" ""
 PREFIX=arm-none-eabi
@@ -261,7 +266,17 @@ endif
 AFLAGS+=${patsubst %,-I%,$(call fixpath,$(IPATH))}
 CFLAGS+=${patsubst %,-I%,$(call fixpath,$(IPATH))}
 CXXFLAGS+=${patsubst %,-I%,$(call fixpath,$(IPATH))}
+ifneq ($(MSYS),)
+# 2-27-2023:  This workaround was added to resolve a linker bug introduced
+# when we started using ln_args.txt.  The GCC linker expects C:/-like paths
+# on Windows if arguments are passed in from a text file.  However, ln_args
+# is parsed through a regex that misses the edge case -L/C/Path/... because
+# of the leading "-L".  We use cygpath here to handle that edge case before
+# parsing ln_args.txt.
+LDFLAGS+=${patsubst %,-L%,$(shell cygpath -m $(LIBPATH))}
+else
 LDFLAGS+=${patsubst %,-L%,$(call fixpath,$(LIBPATH))}
+endif

 ################################################################################
 # The rule for building the object file from each C source file.
```

Jake-Carter, 2023-02-27:
Note for maintainers:  The root cause of the issue is our regex here:  https://github.com/Analog-Devices-MSDK/msdk/blob/1fa83df5386af1a14ba75411e2d1e4ff9ed768a7/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk#L356

This misses our LIBPATH formation here:  https://github.com/Analog-Devices-MSDK/msdk/blob/1fa83df5386af1a14ba75411e2d1e4ff9ed768a7/Libraries/CMSIS/Device/Maxim/MAX32655/Source/GCC/gcc.mk#L264

on MSYS2.  It gets parsed to `-L/C/path...` which breaks the search path.  GCC expects `C:/-like` paths on Windows.

We will need to apply the change above to all gcc.mk files.

PR #479 (2023-03-13): Fix ln_args parsing bug on MSYS2
Resolves #448