analogdevicesinc/msdk #485: Bootload the program through serial UART/I2C

chaitanya-smartrotamac, 2023-03-13:
Tailing to the Issues posted on the Secure-loader Git page

https://github.com/MaximIntegratedAI/secure-loader/issues/3#issue-1618428238
https://github.com/MaximIntegratedAI/secure-loader/issues/4#issue-1621104811

I am completely struck in this issue since a week and I am totally confused.

1)Can we bootload the code to the MAX78000 using  UART/I2C/SPI? Is there a clear documentation regarding that?
2) Can we use the MAXDAP(MAX32625PICO) to use the MAXIM bootloader tools?

I tried to upload the latest dap firmware [max32625_max32666fthr_if_crc_swd_v1.0.4_pico2.bin] from BT/daplink/ but I cant
as the MAXDAP cant enter maintenance mode at all.
I get this 
[image]
some times it enables and sometimes it can't. 
is there a solution for this?

Please someone help me.

Jake-Carter, 2023-03-13:
Hi @chaitanya-smartrotamac, I've been reviewing our documentation.  I agree that it is not clear at all.  I'm tracking down some resources to support you.  Theoretically we can port the [Bootloader_Host](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX32660/Bootloader_Host) example to the MAX78000, but I am confirming internally exactly which version of the bootloader is on the MAX78000.  There are some mismatches in our documentation about that as well.

Apologies for the confusion, I will keep you updated as I uncover some more info.

chaitanya-smartrotamac, 2023-03-15:
HI, @Jake-Carter Thank you for your response.
Is there a possibility that I can get the source code for this bin file?
So, I can develop my own bootloader to communicate with MAX78000?
https://github.com/MaximIntegratedAI/secure-loader/blob/main/MAX78000_Bootloader_v3_4_4.bin

Jake-Carter, 2023-03-16:
@chaitanya-smartrotamac  Sure, I'll provide it if I can track it down.  

Here's what I've been able to clarify so far:  There are _two_ types of bootloaders that we have.

1. The first type are _ROM_-based bootloaders.  On startup, these ROM-based bootloaders look for an activation condition that is documented in the microcontroller's User Guide.  When that condition is met, it offers a serial interface (usually UART or I2C) for loading programs into flash.

2. The other type are _Flash_-based bootloaders.  These usually do _not_ come loaded into the device by default.  They must be loaded over SWD first.  Afterwards, they present their own activation conditions and bootloader interface over I2C/SPI/UART.  The [secure-loader](https://github.com/MaximIntegratedAI/secure-loader) is this type of _flash_-based bootloader.

Additionally, the two types of bootloaders _each_ come in two types of variants:
1. Non-secure (No extra security features)
3. Secure (Validates and protects the flashed application with extra layers of security)

Therefore we have 4 total possibilities:
1. Non-secure _ROM_-based bootloader
2. Secure _ROM_-based bootloader
4. Non-secure _flash_-based bootloader
5. Secure _flash_-based bootloader

From what I can gather from my colleagues the MAX78000 comes with type 1 (Non-secure _ROM_-based bootloader) by default.

Its usage is covered by the [MAX78000 User Guide](https://www.analog.com/media/en/technical-documentation/user-guides/max78000-user-guide.pdf) **Section 26**.  Its activation is covered by Sections 26.4 and 26.5, and the command-set is documented in section 26.8 (general commands).  Any documentation labelled with _secure_ is not applicable for it.  

My colleagues have also provided some host scripts for driving the type 1 bootloader in the MAX78000 from a _host PC_.  They're fairly minimal and undocumented, but I have not been able to find any _microcontroller_ example host code yet.  Hopefully they're a somewhat useful reference until I can pull together some better resources.

* [max78000_rombootloader_host.zip](https://github.com/Analog-Devices-MSDK/msdk/files/10996632/max78000_rombootloader_host.zip)

chaitanya-smartrotamac, 2023-03-17:
Thank you @Jake-Carter for your efforts. 
I will check this https://github.com/Analog-Devices-MSDK/msdk/files/10996632/max78000_rombootloader_host.zip 

I am developing a secondary bootloader to load the code through UART but the code size is pretty huge to fit as a bootloader.
Are there any documents points to the _assembly coding in MAX78000_ and how to _create linker files for the MAX78000_ compiler will be very much useful.

Jake-Carter, 2023-03-17:
The ROM bootloader for the MAX78000 should use UART by default.  I think for your use-case it's probably the best option, since it does not occupy any additional space in flash.

> Are there any documents points to the assembly coding in MAX78000 and how to create linker files for the MAX78000 compiler will be very much useful.

For assembly coding you can refer to the [Cortex-M4 instruction set](https://developer.arm.com/documentation/ddi0439/b/Programmers-Model/Instruction-set-summary/Cortex-M4-instructions?lang=en) documentation from Arm.

To add an assembly file to a build, save it with the `.S` file extension.  You _must_ use capital `.S` for our build system to pick it up.  You'll also need to add it to the `SRCS` and `VPATH` [build configuration variables](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#build-configuration-variables).

Our [startup file](https://github.com/Analog-Devices-MSDK/msdk/blob/main/Libraries/CMSIS/Device/Maxim/MAX78000/Source/GCC/startup_max78000.S) is written in assembly and could be used as an example.

For linker files you can refer to the [GNU ld manual](https://sourceware.org/binutils/docs/ld/Scripts.html) and use our [default linker file](https://github.com/Analog-Devices-MSDK/msdk/blob/main/Libraries/CMSIS/Device/Maxim/MAX78000/Source/GCC/max78000.ld) as an example.  The `LINKERFILE` [build config variable](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#build-configuration-variables) can be used to change the linkerfile for the build.  I haven't really found a great reference for _Arm-specific_ linkerfiles, but there are a few articles online.

Jake-Carter, 2023-05-03:
Hi @chaitanya-smartrotamac, I will close this issue for now.  Please let us know if there's anything else we can help clarify

closed 2023-05-03 by Jake-Carter

fcenciarelli, 2023-08-13:
Hi @Jake-Carter , I am trying to use an external MCU to load code to MAX78000 through UART. 

I can reliably activate the bootloader through UART. I can send the command "i\r\n" to get back "USN:..".
I can also send the load command "l\r\n" and get back the "Ready to load SREC", however I am not sure in what format and timing should I send the data. I get a "Load success..." straight away.
On the PC to MAX78000 Python implementation you shared it seems like the pages must be erased before loading the code?

It would be great to have more documentation on how to reliably load the file, which format to use when loading, and the timing of the communication between MAX78000 and an host MCU, or if you have any implementation in C of the python code you shared.

Thank you in advance!