analogdevicesinc/msdk #505: Update to new version v2023_03_Release cant build any code

chaitanya-smartrotamac, 2023-03-23:
I updated today to [v2023_03_Release](https://github.com/Analog-Devices-MSDK/msdk/releases/tag/v2023_03_Release) I tried to build my codes it is throwing a error in build.
Then I shifted to new workspace expecting something wrong with my workspace, But still error remains same.

Error

```
15:18:52 **** Build of configuration Default for project Hellow ****
make -r -j 8 --output-sync=target --no-print-directory 
Loaded project.mk
/usr/bin/make -f C:/MaximSDK/Libraries/PeriphDrivers/libPeriphDriver.mk  lib BUILD_DIR=C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/softfp PROJ_CFLAGS="-Wall  -DMXC_ASSERT_ENABLE -DCAMERA_OV7692" PROJ_LDFLAGS="" MXC_OPTIMIZE_CFLAGS=-O2 MFLOAT_ABI=softfp DUAL_CORE= RISCV_CORE=
  CC    main.c
make[1]: *** No rule to make target 'c:\maximsdk\libraries\periphdrivers\source\aes\aes_key_revb_regs.h', needed by 'C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/softfp/aes_me17.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make: *** [C:/MaximSDK/Libraries/PeriphDrivers/periphdriver.mk:96: C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/softfp/libPeriphDriver_softfp.a] Error 2
make: *** Waiting for unfinished jobs....
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/../../../MiscDrivers/PushButton/pb.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/../../../MiscDrivers/LED/led.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/Source/board.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/../../../MiscDrivers/stdio.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/../../../MiscDrivers/Camera/camera.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/../../../MiscDrivers/Display/tft_ili9341.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78000/FTHR_RevA/../../../MiscDrivers/Camera/ov7692.c
"make -r -j 8 --output-sync=target --no-print-directory" terminated with exit code 2. Build might be incomplete.
```

I don't know why it is looking for this file which I don't have in my system 
-- > c:\maximsdk\libraries\periphdrivers\source\aes\aes_key_revb_regs.h

chaitanya-smartrotamac, 2023-03-23:
After that
I changed the floating point settings
```
# Set hardware floating point acceleration.
# Options are:
# - hard
# - soft
# - softfp (default if MFLOAT_ABI is not set)
MFLOAT_ABI ?= hard
```

Now the errors increased
```
15:39:55 **** Incremental Build of configuration Default for project Hellow ****
make -r -j 8 --output-sync=target --no-print-directory 
Loaded project.mk
/usr/bin/make -f C:/MaximSDK/Libraries/PeriphDrivers/libPeriphDriver.mk  lib BUILD_DIR=C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/hardfp PROJ_CFLAGS="-Wall  -DMXC_ASSERT_ENABLE -DCAMERA_OV7692" PROJ_LDFLAGS="" MXC_OPTIMIZE_CFLAGS=-O2 MFLOAT_ABI=hard DUAL_CORE= RISCV_CORE=
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/AES/aes_me17.c
make[1]: *** Waiting for unfinished jobs....
In file included from C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/AES/aes_me17.c:43:
c:\maximsdk\libraries\periphdrivers\source\aes\aes_revb.h:40:10: fatal error: aeskeys_revb_regs.h: No such file or directory
   40 | #include "aeskeys_revb_regs.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [C:/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX78000/Source/GCC/gcc.mk:282: C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/hardfp/aes_me17.o] Error 1
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/AES/aes_revb.c
In file included from C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/AES/aes_revb.c:45:
c:\maximsdk\libraries\periphdrivers\source\aes\aes_revb.h:40:10: fatal error: aeskeys_revb_regs.h: No such file or directory
   40 | #include "aeskeys_revb_regs.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [C:/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX78000/Source/GCC/gcc.mk:282: C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/hardfp/aes_revb.o] Error 1
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/GPIO/gpio_common.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/GPIO/gpio_reva.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/GPIO/gpio_ai85.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/I2C/i2c_me17.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/I2S/i2s_ai85.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/I2C/i2c_reva.c
make: *** [C:/MaximSDK/Libraries/PeriphDrivers/periphdriver.mk:96: C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78000/hardfp/libPeriphDriver_hardfp.a] Error 2
"make -r -j 8 --output-sync=target --no-print-directory" terminated with exit code 2. Build might be incomplete.

15:39:58 Build Failed. 6 errors, 0 warnings. (took 2s.454ms)
```

Everything is working absolutely fine till yesterday.

chaitanya-smartrotamac, 2023-03-23:
Uninstalling everything and reinstalling solved the issue.

Jake-Carter, 2023-03-24:
Hi @chaitanya-smartrotamac, this is most likely caused by an outdated peripheral driver library file.  If you had a project with an older library file but then attempted to build and link a _newer_ project on top of it, you could get the object file mismatches shown above.

For future reference, you can run the `make clean.periph` command (or [build task](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#building-a-project)) and it will completely clean old peripheral driver library builds.  On the next build, it will be recreated from scratch.

Were you working from a copy of the Github repo, or a copy of the release installation?  The release is supposed to redistribute new peripheral driver library files to address this issue, and should have wiped the `Libraries/PeriphDrivers/bin/` folders for each micro on the update.  Maybe there is a bug in our installer packaging.  I'll take a look

Jake-Carter, 2023-03-27:
Replicated the issue.  Attempting to link a Hello_World project against the updated lib files.

A suspected bug in the installer packaging is under investigation.

**For those who encounter this issue:  run `make clean.periph` or the [build task](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#building-a-project)** to rebuild the Peripheral Library from scratch on the next build.

```Shell
make -r -j 8 --output-sync=target --no-print-directory TARGET=MAX78002 BOARD=EvKit_V1 MAXIM_PATH=C:/MaximSDK MAKE=make PROJECT=Hello_World 

Loaded project.mk
make -f C:/MaximSDK/Libraries/PeriphDrivers/libPeriphDriver.mk 
 lib BUILD_DIR=C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78002/softfp PROJ_CFLAGS="-Wall  -DMXC_ASSERT_ENABLE -fdiagnostics-color=always -DTFT_NEWHAVEN -DFLIP_SCREEN -DSWAP_XY -DCAMERA_OV7692" PROJ_LDFLAGS="" MXC_OPTIMIZE_CFLAGS=-O2 MFLOAT_ABI=softfp 
DUAL_CORE= RISCV_CORE=
  CC    main.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/LED/led.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/Source/board.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/PushButton/pb.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/Touchscreen/tsc2046.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/Camera/mipi_camera.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/Camera/camera.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/Display/tft_st7789v.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/Camera/ov7692.c
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/Camera/sccb.c
  AS    C:/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX78002/Source/GCC/startup_max78002.S
  CC    C:/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX78002/Source/heap.c
  CC    C:/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX78002/Source/system_max78002.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/SYS/mxc_assert.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/SYS/mxc_lock.c
make[1]: *** No rule to make target 'c:\maximsdk\libraries\periphdrivers\source\aes\aes_key_revb_regs.h', needed by 'C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78002/softfp/aes_ai87.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/SYS/mxc_delay.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/SYS/pins_ai87.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/SYS/nvic_table.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/ADC/adc_ai87.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/SYS/sys_ai87.c
  CC    C:/MaximSDK/Libraries/CMSIS/../PeriphDrivers/Source/ADC/adc_revb.c
make: *** [C:/MaximSDK/Libraries/PeriphDrivers/periphdriver.mk:96: C:/MaximSDK/Libraries/PeriphDrivers/bin/MAX78002/softfp/libPeriphDriver_softfp.a] Error 2
make: *** Waiting for unfinished jobs....
  CC    C:/MaximSDK/Libraries/Boards/MAX78002/EvKit_V1/../../../MiscDrivers/stdio.c
```

Jake-Carter, 2023-03-27:
Note for maintainers:  As of https://github.com/Analog-Devices-MSDK/msdk/pull/457 this should not be an issue.  However, our Jenkins builders package the static libs with a separate method.

Jake-Carter, 2023-06-30:
As of the [June 2023](https://github.com/Analog-Devices-MSDK/msdk/releases/tag/v2023_06) release this issue should be resolved.

Our installer now removes the problematic old dependency files left over from previous builds when it updates the PeriphDrivers component.  (https://github.com/Analog-Devices-MSDK/Packages/commit/a1bafdeec5ff584aeb15c916fe3bbe1388545758, https://github.com/Analog-Devices-MSDK/Packages/commit/c3b30f9f84922279331a25522f4ca8ab0814bf01)

closed 2023-06-30 by Jake-Carter