analogdevicesinc/msdk #507: Documentation for Keil/IAR

jlj-ee, 2023-03-24:
Hi, I am working with the MAX32670, initially via the EvKit but soon in a custom board.

I've been very impressed with the support for VS Code integration and ease of getting running, both with the examples and with new code via the "New_Project" and "Inject" resources -- thanks for that! However, I may have to make use of Keil or IAR due to external pressures, and the MSDK documentation there was limited to the extent that I was questioning whether they are still officially-supported IDEs.

**Are Keil/IAR still considered supported by the MSDK? If so, could the [User Guide](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#supported-development-environments) be updated and a boilerplate project added for those IDEs, if such a resource does not already exist?**

I tried to do my due diligence and searched through the MSDK repo for references to Keil/IAR, but did not see any sample projects, at least for the MAX32670. I was able to get a CMSIS pack (dated 2021?) in the Keil uVision 5 pack installer and found projects in the examples tab there, though I am not sure if they are up-to-date given that there have been more recent MSDK releases.

My plan was to adapt one of those Keil example projects to for my new code, in the absence of a better boilerplate option.

Apologies if I missed something and all of my questions are already answered somewhere. Thanks again for these great development resources!

jlj-ee, 2023-03-24:
It appears that this may extend beyond just a missing documentation issue. Actually attempting to build the most basic example project in the Keil pack (Hello_World) with CMSIS 5.6.0 (which the MAX32670 pack prompted me to install) results in build errors:

`C:\Keil_v5\ARM\PACK\Maxim\MAX32670\1.0.3\Libraries\PeriphDrivers\Include\MAX32670\nvic_table.h(54): error:  #147-D: declaration is incompatible with "void __NVIC_SetVector(IRQn_Type, uint32_t)"  
(declared at line 1906 of "C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.6.0\CMSIS\Core\Include\core_cm4.h")
void NVIC_SetVector (IRQn_Type irqn, void (*irq_callback) (void));`

Jake-Carter, 2023-03-24:
Hi @jlj-ee, thank you for your feedback and for reporting the issue.

Keil and IAR are still officially supported by the MSDK.  This type of question is now one we've gotten a few times, so I'll add some info to the UG as you suggested.  Everything is distributed through the pack files, so we have been trying to lean on the official Keil/IAR docs as much as possible.  I agree at the very least we should have some basic info in our UG.

In this case, it looks like the pack files have become quite outdated in relation to the CMSIS library.  We've addressed this for our GCC-based projects (VS Code/Eclipse) but not IAR/Keil.  I've tagged our maintainer @ftariqAnalogTx and we'll get back to you in more detail early next week regarding the pack files and build error

jlj-ee, 2023-03-26:
Thank you @Jake-Carter for the follow-up!
I definitely understand offloading some documentation load via existing Keil/IAR docs; maybe just some links to relevant pages would help?

And thanks @ftariqAnalogTx in advance for looking into the outdated pack files! Let me know if there is any other info I can provide that will help. In the short term I can continue my development with my VS Code project.

jlj-ee, 2023-03-27:
Hi @ftariqAnalogTx 

Thanks for the suggestion -- I tried the changes you suggested and was able to build an example. Obviously not ideal to be editing files outside the MAX32670 pack though, since it could affect working with other packs/targets. Assuming this overall build issue will be fixed in an updated MAX32670 pack, I will simply stick to GCC-based project for now.

Also, just wanted to note that there are also 3 minor BSP-related warnings that also appear when building the pack version but not the GCC projects:

1. `C:/Keil_v5/ARM/PACK/Maxim/MAX32670/1.0.3/Libraries/Boards/MAX32670/Source/pb.c(70): warning:  #188-D: enumerated type mixed with another type` 
`  NVIC_EnableIRQ(MXC_GPIO_GET_IRQ(MXC_GPIO_GET_IDX(pb_pin[pb].port)));`
2. `C:/Keil_v5/ARM/PACK/Maxim/MAX32670/1.0.3/Libraries/Boards/MAX32670/Source/led.c(85): warning:  #1-D: last line of file ends without a newline`
`  }`
3. `C:/Keil_v5/ARM/PACK/Maxim/MAX32670/1.0.3/Libraries/PeriphDrivers/Source/SYS/pins_me15.c(92): warning:  #1-D: last line of file ends without a newline`
`  const mxc_gpio_cfg_t gpio_cfg_chrg_lc2      =    { MXC_GPIO0, MXC_GPIO_PIN_17, MXC_GPIO_FUNC_ALT4, MXC_GPIO_PAD_NONE };`

I am fine ignoring/fixing these locally for now but just wanted to log them here.

Jake-Carter, 2023-03-31:
Re-opening to track the outdated pack file

Jake-Carter, 2023-06-08:
Hi @jlj-ee, just following up on this older ticket.  We've had the pack file updated, but Arm is having some trouble on their side getting their distribution link to work.

For you and other users in the meantime the pack file is available from the link below:

https://mxim.net/microcontroller/pack/Maxim.MAX32670.1.1.0.pack

jlj-ee, 2023-07-01:
Thanks @Jake-Carter, the update on this issue is much appreciated. I'm ok with closing this or leaving it open until the ARM distribution link issue is resolved. Are there any plans to keep the CMSIS pack and ARMCC MSDK releases handled by this GitHub repo more in sync?

Jake-Carter, 2023-07-10:
Thanks @jlj-ee, I'll leave it open since until the link is working (https://www.keil.arm.com/packs/max32670-maxim/)

@ftariqAnalogTx is working on keeping the packs synced up.  We'll keep following up with the Keil folks to get the Keil links working asap

Jake-Carter, 2023-08-10:
@jlj-ee Arm's got the pack files uploaded now, and the link is working (https://www.keil.arm.com/packs/max32670-maxim/boards/), so I'll close this out

closed 2023-08-10 by Jake-Carter