analogdevicesinc/msdk #513: MAX32672 Evaluation Kit - LP Example - Current Consumption

leoreyesth, 2023-03-27:
Hi,

I'm trying to achieve the low power consumption of the MAX32672 in the EV kit.

I uploaded the LP example code and followed the steps from the EV Kit Datasheet to measure the current.

The lowest measured power consumption at deepsleep mode was 470uA @3.3V at VDD, but the DataSheet states 4.4uA @3.3V at VDD.

Is there some  modification to the LP example code to lower even more the power consumption?

Jacob-Scheiffler, 2023-03-28:
@leoreyesth Which rail are you measuring current on?

Jacob-Scheiffler, 2023-03-30:
@leoreyesth Sorry for the delay, but I got it figured out this morning. The steps I followed were:
- Remove jumpers JP2 through JP11
- Disconnect LCD
- Set all GPIO pins low except for SWD pins (P0.0/P0.1), push button (P0.18), and P0.10.
- Set GPIO pin P0.10 high. (It is connected to a pullup resistor that can't be disabled.)

This the function I used to set the GPIO pins to the appropriate state:
```
#include "gpio.h"
void configure_gpios(void)
{
	mxc_gpio_cfg_t out_clr;
	mxc_gpio_cfg_t out_set;

	// Set all GPIO pins low except for SWD, push button, and P0.10
	out_clr.port = MXC_GPIO0;
	out_clr.mask = 0xFFFBFBFC;
	out_clr.func = MXC_GPIO_FUNC_OUT;
	out_clr.pad = MXC_GPIO_PAD_NONE;
	out_clr.vssel = MXC_GPIO_VSSEL_VDDIOH;
	MXC_GPIO_Config(&out_clr);
	MXC_GPIO_OutClr(out_clr.port, out_clr.mask);

	//Set GPIO P0.10 high
	out_set.port = MXC_GPIO0;
	out_set.mask = MXC_GPIO_PIN_10;
	out_set.func = MXC_GPIO_FUNC_OUT;
	out_set.pad = MXC_GPIO_PAD_NONE;
	out_set.vssel = MXC_GPIO_VSSEL_VDDIOH;
	MXC_GPIO_Config(&out_set);
	MXC_GPIO_OutSet(out_set.port, out_set.mask);
}
```

With these configurations I was able to get a Deep Sleep current of 3.1uA and Backup current of 0.62uA. Also, you may have had a typo, but the VDD jumpers is actually JP13. Hope this helps!

leoreyesth, 2023-03-30:
Thanks @Jacob-Scheiffler. I will try the example.
I dont know if a real problem but the function MXC_GPIO_Confing sets the padctrl1 register that is reserved and read only, and also sets the vssel register that is reserved DN

Jake-Carter, 2023-03-30:
Closing issue as #519 is merged.  GPIO drivers will be modified in a separate PR

closed 2023-03-30 by Jake-Carter