analogdevicesinc/msdk #516: MSDK installation erros on Ubuntu 22.04

plat-zhaw, 2023-03-28:
Hi there,

I downloaded and installed the Maxim Micros SDK on Ubuntu 22.04. I got two different errors.

The first error occurs when the installer tries to make the directory for the links on the desktop. I simply ignored the first message `Error: Operation MkDir does not exist.`. The second message `Error during installation process (com.maximintegrated.intaller): Execution failed (Unexpected exit code: 1) "ln -s /home/plat/MaximSDK /home/plat/Desktop/MaximSDK/MaximSDK"` shows the path of the missing directory. Is simply created the directory `/home/plat/Desktop/MaximSDK/MaximSDK` manually. Then I pressed the button `Retry` and it created the link. 

[image]
[image]

The second error is about the Eclipse application launcher. The error messages below show that the installer tries to acces the file `/usr/share/applications/Eclipse-MaximSDK.desktop`. So I went to the the parent folder to try and assign the file the correct permission as shown in the second error message, but there is no such file in there, as the bash transcript below shows.

```bash
plat@6674:/usr/share/applications$ ls -al | grep -i maxim
plat@6674:/usr/share/applications$ ls -al | grep -i eclipse
plat@6674:/usr/share/applications$ 
```

[image]
[image]

I could solve the first error. The second error just means that the application launcher for the Eclipse application is missing. I do not plan to use this. I much rather use VS Code anyway. I just hope there will be no problems down the line.

Kind regards.

Jake-Carter, 2023-03-28:
Hi @plat-zhaw, thank you very much for reporting this.
* Error 1:  This is a typo in our installer package.  It should be `Mkdir` instead of `MkDir`.
* Error 2: The installer needs elevated permissions to write to `/usr/share/...`, but we are missing a call to elevate this operation.

I'll be pushing fixes to our installer package to address these bugs.  You're right, the SDK should be fully functional despite the missing Eclipse shortcut.

plat-zhaw, 2023-03-29:
Hi @Jake-Carter, thanks for the fast answer. I am glad that the SDK should be usable. That is the most important thing for me.

Jake-Carter, 2023-04-20:
Updated component files have now been pushed to the release server.  As of 4/20/2023 this issue should be fixed.

Thanks again for reporting @plat-zhaw

closed 2023-04-20 by Jake-Carter