analogdevicesinc/msdk #518: MAX32670 MXC_SYS clock configuration

jlj-ee, 2023-03-28:
Hi, is there a bug in the MXC_SYS implementation for the MAX32670?

In the [UG](https://www.analog.com/media/en/technical-documentation/user-guides/max32670max32671-user-guide.pdf), the clock tree (page 38) and GCR_CLKCTRL register description (page 66-67) both document an external clock as an available system clock option.
However, the implementation in sys_me15.c has support for EXTCLK removed:

In MXC_SYS_ClockSourceEnable (returns an error)
```C
    case MXC_SYS_CLOCK_EXTCLK:
        // MXC_GCR->clkctrl |= MXC_F_GCR_CLKCTRL_EXTCLK_EN;
        // return MXC_SYS_Clock_Timeout(MXC_F_GCR_CLKCTRL_EXTCLK_RDY);
        return E_NOT_SUPPORTED;
        break;
```

In MXC_SYS_ClockSourceDisable (silent failure)
```C
    case MXC_SYS_CLOCK_EXTCLK:
        // MXC_GCR->clkctrl &= ~MXC_F_GCR_CLKCTRL_EXTCLK_EN;
        break;
```

In MXC_SYS_ClockSelect (silent failure)
```C
    case MXC_SYS_CLOCK_EXTCLK:
        // Enable HIRC clock
        // if(!(MXC_GCR->clkctrl & MXC_F_GCR_CLKCTRL_EXTCLK_EN)) {
        //     MXC_GCR->clkctrl |=MXC_F_GCR_CLKCTRL_EXTCLK_EN;

        //     // Check if HIRC clock is ready
        //     if (MXC_SYS_Clock_Timeout(MXC_F_GCR_CLKCTRL_EXTCLK_RDY) != E_NO_ERROR) {
        //         return E_TIME_OUT;
        //     }
        // }

        // Set HIRC clock as System Clock
        // MXC_SETFIELD(MXC_GCR->clkctrl, MXC_F_GCR_CLKCTRL_SYSCLK_SEL, MXC_S_GCR_CLKCTRL_SYSCLK_SEL_EXTCLK);

        break;
```

I will be populating the EXT_CLK footprint on my EV kit to test this disabled code.

(Also, MXC_SYS does not appear in the [peripheral documentation](https://analog-devices-msdk.github.io/msdk/Libraries/PeriphDrivers/Documentation/MAX32670/modules.html).)

jlj-ee, 2023-03-28:
Thanks @Jake-Carter for looking into this! That sequence for establishing the EXT_CLK input makes sense.
I am also happy to help test on my EV Kit.

Jake-Carter, 2023-04-04:
Hi @jlj-ee, I added EXT_CLK support for the MAX32670 and tested it at 2Mhz.  I don't have a clean enough connection to test at higher speeds, but it looks like `Hello_World` is working as expected off of the external input.

See the [`EXT_CLK`](https://github.com/Jake-Carter/msdk/tree/fix/gh-518/Examples/MAX32670/EXT_CLK) example on my fork.

I'll be porting this to the other micros as well before merging the PR.

jlj-ee, 2023-04-06:
Hi @Jake-Carter, I got a chance to verify the example from your fork on my EVKit and it looks great!
When I have some more time I will look at additional speeds. Thanks for getting this fix in!

PR #521 (2023-04-12): MXC_SYS:  Add Support for External Clock Inputs, Fix MXC_SYS API Documentation
Resolves #518 

## Overvew

- Enable the external clock input as an option for the system clock source for microcontrollers that support it (mxc_sys).
- Add the `EXT_CLK` example for the MAX32670.
- Update Doxygen files to include static members.  This fixes missing documentation for some mxc_sys functions (`MXC_SYS_Crit_xxx`) and macro (`MXC_CRITICAL(xxx)`).
- Fix missing mxc_sys section in API doxygen for all micros.
- Improve mxc_sys API documentation (see [here ](https://jake-carter.github.io/msdk/Libraries/Peri [... cut]
closed 2023-04-12 by Jake-Carter