analogdevicesinc/msdk #542: MSDK Installer Failure

EricB-ADI, 2023-04-08:
I was installing all of our tools on a new install of ubuntu using the MSDK installer, and it seems that there is an issue with one of the commands. it looks like it used ``MkDir`` instead of ``mkdir`` in the shell. Second issue seems to be with a bad symlink.

[image]
[image]

[image]

[image]

Jake-Carter, 2023-04-08:
Thanks @EricB-ADI, this is the same issue as #516 

I've had a fix to the installer package staged for release on the test server.  Just waiting for approval to push it out to production next week.

Agreed we need some auto-testing of the installer.  It's on my task list, should be able to get to it soon

EricB-ADI, 2023-04-10:
Alright, im gonna close this then

closed 2023-04-10 by EricB-ADI