analogdevicesinc/msdk #566: PR Check Actions Not Operating as Intended

jdk-maxim, 2023-04-25:
For recent pull requests, I found that several of the required Check Actions are being skipped incorrectly.

### Build_Examples
[This fails](https://github.com/Analog-Devices-MSDK/msdk/actions/runs/4798855615/jobs/8537699918?pr=565) on recent pull request, but passed on [virtually identical one](https://github.com/Analog-Devices-MSDK/msdk/pull/549) from last week.  I noticed that last weeks just skipped running this test completely.

Looking through list of recent PRs, several appear to skip the actual build run, after "Check watch files" stage, due to "$RUN_TEST" never being set. For example:
[dtm sweep with better plotting](https://github.com/Analog-Devices-MSDK/msdk/actions/runs/4755259075/jobs/8449167735?pr=560)
[MAX32570 cameraif dma bit index updates, tested it works ](https://github.com/Analog-Devices-MSDK/msdk/actions/runs/4744210502/jobs/8424808562?pr=556)

Guessing this section might fail due to case sensitivity:
```
  # Assume we want to actually run the workflow if no files changed
  for watch_file in $WATCH_FILES; do 
    if [[ "$CHANGE_FILES" == *"$watch_file" ]]; then
      RUN_TEST=1
    fi
  done
```

### Lint Code Base
[This passed ](https://github.com/Analog-Devices-MSDK/msdk/actions/runs/4798855606/jobs/8537699891?pr=565) but didn't actually test anything.  Log shows:
```
 fatal: ambiguous argument 'master...6732539c3fa07fc82c54e01ccc3976fc862b8405': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
2023-04-25 14:45:52 [WARN]   No files were found in the GITHUB_WORKSPACE:[/github/workspace] to lint!
```
But this did not get bubbled up to to the top for a Red Failing checking.  Maybe this failed since we migrated from master->main?

### Verify Register and SVD Files
[This doesn't get run](https://github.com/Analog-Devices-MSDK/msdk/actions/runs/4798855603/jobs/8537699913?pr=565) even though the PR contains changes to SVD and some "*_regs.h" files.

sihyung-maxim, 2023-05-01:
1. There have been some changes to the Build_Examples and BLE_Examples workflows recently which have caused those workflows to fail occasionally in the past few weeks. I believe the changes have been (mostly) finalized.

2. I've updated the linter's default branch to main instead of master.

3. Something in the Verify_Register_SVD's workspace is preventing `git diff` from working. I'm taking a look at it now.

sihyung-maxim, 2023-05-01:
The Verify_Register_SVD workflow checkouts the msdk twice if the source branch of the PR is from a forked branch. This resulted in the `git diff` returning nothing even if there are file changes. I updated the workflow to make sure it only checks out the branch once.

All the changes were merged with PR: #565.

Jake-Carter, 2023-05-03:
Thanks @sihyung-maxim 

In #545 I fixed some issues but didn't modify the bash script that checks for watch files that Jeremy pointed out above.

Keeping an eye on things...  We may need to reopen this to reimplement the check if it continues to fail intermittently after your latest fixes

Jake-Carter, 2023-05-03:
Register file changes in #534 did not trigger build_examples

sihyung-maxim, 2023-05-05:
Whoops, did not know merging a PR with a linked issue would automatically close the issue. My bad @Jake

PR #577 (2023-05-05): Build Workflow:  Update Check Watch Files Implementation
The old "check watch files" implementation seemed to have some issues with case sensitivity.  The new implementation from @jdk-maxim seems to be much more reliable.

Fixes #566 

Manual run on fork works great:  https://github.com/Jake-Carter/msdk/actions/runs/4886879634/jobs/8722797077

This PR should trigger a build based on its test commit as well.
closed 2023-05-05 by Jake-Carter