analogdevicesinc/msdk #593: MAX32690 : Unable to generate SPI Clock > 15 MHz

ConWalsh, 2023-05-25:
Using the MAX32690 EVK, an attempt was made to carry out an SPI transaction for a given SPI_SPEED. It was observed that this breaks for a frequency greater than 15 MHz. It is not an issue due to the loopback configuration in the example (can be replicated when not shorting MOSI-MISO). SPI bus is connected with 0 Ohm links to a codec on the EVK. These were removed to verify that this was not causing the issue. Clock is being correctly set by the BSP to the IPO and this was verified by looking at the GCR register. No error codes are returned at any point.

See Example used to test:
`/***** Includes *****/
#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
#include <string.h>
#include "board.h"
#include "mxc_device.h"
#include "mxc_delay.h"
#include "mxc_pins.h"
#include "nvic_table.h"
#include "uart.h"
#include "spi.h"
#include "dma.h"
#include "led.h"

#include "mxc_sys.h"

/***** Definitions *****/
#define DATA_LEN 1 // Words

// Testing
//#define SPI_SPEED 15000000      //  Works okay
#define SPI_SPEED 20000000 	//  Not seeing clock for SPI_SPEED > 15 MHz
//#define SPI_SPEED 30000000    //

#define DATA_VALUE_8BIT 0xA5         // 8-bit
#define DATA_VALUE_16BIT 0xA5A5      // 16-bits
#define DATA_VALUE_32BIT 0xA5A5A5A5  // 32-bits

#define SPI MXC_SPI0
#define SPI_IRQ SPI0_IRQn
#define MOSI_PIN 28
#define MISO_PIN 27

/***** Globals *****/

volatile int SPI_FLAG;
volatile uint8_t DMA_FLAG = 0;

// Data Buffers
// 8-bit
uint8_t rx_data_8_bit[DATA_LEN];
uint8_t tx_data_8_bit[DATA_LEN];

// 16-bit
uint16_t rx_data_16_bit[DATA_LEN];
uint16_t tx_data_16_bit[DATA_LEN];

// 32-bit
uint32_t rx_data_32_bit[DATA_LEN];
uint32_t tx_data_32_bit[DATA_LEN];

int main(void)
{`
`

	volatile int retVal;

	printf("\n**************************** SPI MASTER TEST *************************\n");

	// Ensure the System Clock uses the 120 MHz internal primary oscillator (IPO)
	// Already set in SystemInit()

	// Structures for defining pins and parameters
    mxc_spi_req_t req;
    mxc_spi_pins_t spi_pins;

    // Number of clock pulses - use appropriate size for # of bits being sent
    int data_size = 8;
    //int data_size = 16;
    //int data_size = 32;

    // Define Chip Select Pins
    spi_pins.ss0 = FALSE;
    spi_pins.ss1 = TRUE;
    spi_pins.ss2 = FALSE;

    // Setting the data of the transmit data buffer - only 1 element
    // Do this in a loop for larger arrays
    tx_data_8_bit[0] = DATA_VALUE_8BIT;
    //tx_data_16_bit[0] = DATA_VALUE_16BIT;
    //tx_data_32_bit[0] = DATA_VALUE_32BIT;

    printf("\nInitializing SPI\n");

	// Configure the peripheral
	retVal = MXC_SPI_Init(SPI, 1, 0, 1, 0, SPI_SPEED, spi_pins);

    if (retVal != E_NO_ERROR) {
        printf("\nSPI INITIALIZATION ERROR\n");
        return retVal;
    }

    printf("\nSetting SPI Mode\n");

	// Set SPI mode
	MXC_SPI_SetMode(SPI, SPI_MODE_0);

	//SPI Request
	req.spi = SPI;

	req.txData = (uint8_t *)tx_data_8_bit;
	req.rxData = (uint8_t *)rx_data_8_bit;

	//req.txData = (uint8_t *)tx_data_16_bit;
	//req.rxData = (uint8_t *)rx_data_16_bit;

	//req.txData = (uint8_t *)tx_data_32_bit;
	//req.rxData = (uint8_t *)rx_data_32_bit;

	req.txLen = DATA_LEN;
	req.rxLen = DATA_LEN;
	req.ssIdx = 1;
	req.ssDeassert = 1;
	req.txCnt = 0;
	req.rxCnt = 0;
	//req.completeCB = (spi_complete_cb_t)SPI_Callback;
	SPI_FLAG = 1;

    printf("\nSetting Data Size\n");

	// Setting the clock pulses
    MXC_SPI_SetDataSize(SPI, data_size);

    printf("\nSetting SPI Width\n");

    // Setting full/half-duplex
    MXC_SPI_SetWidth(SPI, SPI_WIDTH_STANDARD);
    //MXC_SPI_SetWidth(SPI, SPI_WIDTH_3WIRE);

    printf("\nCarry out a Transaction\n");

    // Carry out a SPI transaction
	MXC_SPI_MasterTransaction(&req);

	//while(1){};
}
`

Jake-Carter, 2023-05-25:
Hi @ConWalsh, also try increasing the drive strength for the GPIO pins to maximum.  This will help get sharp edges at higher clock speeds

ConWalsh, 2023-05-29:
After looking into it further I made an error on my part. I thought I was using VDDIOH (3V3) for the SPI pins but I was using VDDIO (1V8). 
 Including the line:
`MXC_GPIO_SetVSSEL(MXC_GPIO1, MXC_GPIO_VSSEL_VDDIOH, (MXC_GPIO_PIN_23 | MXC_GPIO_PIN_26 | MXC_GPIO_PIN_28 | MXC_GPIO_PIN_29));`
solved my issue.

@sanjayjaroli thank you for that. This will be important for me when I receive my boards as I will use SPI 3 & 4 on the AHB bus.
@Jake-Carter this will be helpful as well. Thanks for the suggestion.

closed 2023-05-29 by ConWalsh

Jake-Carter, 2023-05-30:
No problem, glad you got it working!