analogdevicesinc/msdk #601: Verify_Register_SVD check not working correctly

jdk-maxim, 2023-05-31:
Current PR: https://github.com/Analog-Devices-MSDK/msdk/pull/600 is failing this check.

It appears to have correctly detected the affected registers and SVD files, and found one I originally missed ai85 uart_regs.h.

But the diff is failing.  Pretty sure these are correct in both the SVD and register files.  Maybe this is failing due to it being a PR from a Fork again?

Jake-Carter, 2023-06-27:
@jdk-maxim last I heard this was related to space issues on our local runners. 

@sihyung-maxim any updates on the SVD workflow lately?

Jake-Carter, 2023-07-07:
@sihyung-maxim this workflow seems to also now be broken by our move to Github Enterprise (msdk internal checkout).  See https://github.com/Analog-Devices-MSDK/msdk/actions/runs/5481437740/jobs/9985735026?pr=653