analogdevicesinc/msdk #614: imgCapture Stream "Image received failed, Only received 3/247808"

ryalberti, 2023-06-09:
I have imageCapture set up on my MAX78000FTHR. I run the following make command with these expected results. 
[image]

When I replicate the demo and run 'stream', I receive the error "Image receive failed! Only received 3/247808 bytes". The board is unable to receive any commands. Here you can see I attempted 'reset' which did not give any output, so I exited the program. When connecting again after pressing the reset button on the board, the program runs fine. I can 'capture' and save an image without issue and the image is not distorted in any way. The same was true when compiling and loading from a different computer and at different resolutions. When I was looking for the issue, I found that it was getting stuck in the transmit loop. In some of my earlier tests, it was stuck at byte 3 and would have to be manually reset to resume normal use. I have since reinstalled the SDK and do not have screenshots. 
[image]

In my current testing, I am receiving a different error. Here, I am receiving the following errors. 
Code and output as shown: 
[image]
The resulting image is below. This makes sense since the console never says the image is finished. In repeated tests, the resulting streamed picture looks the same every time. I believe it is having an issue in the same location. 
[image]
The waiting goes on for at least five minutes. When pressing the reset button on the board, I receive the printouts shown in the code above, but stuck at byte 97. I believe those are delayed outputs before I reset the board, which is why it returns to trying to sync. It does not successfully establish communication with the host until I press reset on the board again (shown). 
[image]

Is this an issue related to my building of the program or my SDK install? 
Thank you in advance!

Jake-Carter, 2023-06-09:
Hi @ryalberti, thank you for your report.

This issue should be resolved as of #583, and it has been merged into the main branch of the Github repo.

The specific commit with the fix is https://github.com/Analog-Devices-MSDK/msdk/pull/583/commits/62a4b8e55f970364002fbdcb19171626bb4b889a.

We have some instructions for [developing from the repo](https://github.com/Analog-Devices-MSDK/msdk#developing-from-the-repo), or you can apply the change manually to your project.  The fix will be published to the installer in our next SDK release by the end of this month.

Apologies for the bug!  Let me know if you have any other issues.

ryalberti, 2023-06-12:
That works, thank you so much!

Jake-Carter, 2023-06-12:
Happy to help, thanks for confirming!  I'll close this out.