analogdevicesinc/msdk #616: MAX78000 support of external SPIRAM

matti122, 2023-06-12:
I want to use an external SPI RAM with MAX78000 on a custom PCB.
But I'm not able to find the Driver and example to use the external SPIRAM N01S830HAT22I which is used on MAX78000FTHR Board.
Could you please help to identify how I can use the external SPIRAM.
Thanks

BitSalat, 2023-06-14:
Would be interested as well - on the feather EVK an OnSemi 1MByte SPI RAM is implemented but no library or reference example in the software is available Can u please clarify?

Jake-Carter, 2023-06-15:
You're right, it looks like we're missing drivers for it in the SDK.

The IC looks very similar to the [APS6404 SRAM](https://www.apmemory.com/wp-content/uploads/APM_PSRAM_QSPI-APS6404L-3SQR-v2.3-PKG.pdf) which I recently wrote drivers for.  You can find them in the [QSPI Example](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX78002/QSPI) for the MAX78002.

It should be easy to adapt the example/drivers.  I'll try to do it by the end of the week.  In the meantime you can reference the QSPI example and its drivers yourself

Jake-Carter, 2023-06-16:
@matti122 @BitSalat 

I was able to complete some preliminary drivers this afternoon.  The main challenge for this SRAM is its "hold" feature.  I found I had to disable it in the mode register to get it to work reliably.

You can find the updated drivers and example on the `dev/N01S830HA-drivers` branch of my fork [here](https://github.com/Jake-Carter/msdk/tree/dev/N01S830HA-drivers/Examples/MAX78000/QSPI). 

Again, be warned this is a _preliminary_ solution.  It may change significantly before the next release.  We are trying to freeze our codebase next Tuesday, so this may or may not make it in to the official June release.

Everything seems functional at its max speed of 20Mhz except for tests 7-9, which test operations across page boundaries.  I will need to troubleshoot this further next week.

```terminal
QSPI SRAM Test:
        Test Address: 0x0
        Test size: 640 bytes
        Test count: 480 rows
        Test speed: 2000000 Hz
(Benchmark) Wrote 640 bytes to internal SRAM in 5us
Test 1: Standard SPI write...
        Done (640 bytes in 2594us)
Test 2: Validate w/ standard SPI...
        Read finished (640 bytes in 2593us)
        Checking for mismatches...
        Done
Test 3: Validate w/ QSPI...
        Read finished (640 bytes in 660us)
        Checking for mismatches...
        Done
Test 4: QSPI Write...
        Done (640 bytes in 663us)
Test 5: Validate w/ standard SPI...
        Read finished (640 bytes in 2593us)
        Checking for mismatches...
        Done
Test 6: Validate w/ QSPI...
        Read finished (640 bytes in 660us)
        Checking for mismatches...
        Done
Test 7: QSPI Writing across page boundaries...
        Wrote 307200 bytes in 317693us
Test 8: Validating with standard SPI...
        Failed (305760) mismatches
Test 9: Validating with QSPI...
        Failed (305760) mismatches

Failed with 611520 mismatches (199.06%)!
```

Jake-Carter, 2023-06-27:
@matti122 @BitSalat this did not make the cut-off for our June release, but I'll get in some more polished drivers asap.

It would be helpful if you can confirm you can replicate the linked example above on your boards.

matti122, 2023-06-27:
Hello Jake,
we are still in Design phase and our own PCB is currently not ready.
I tested it with MAX78000 Feather Board and got same results.
Best regards
Matthias

PR #802 (2023-12-22): feat(MiscDrivers): Add N01S830HA Drivers for MAX78000FTHR Board
## Pull Request Template

### Description

Closes #616 

Adds drivers for the N01S830HA on-board the MAX78000FTHR to `Libraries/MiscDrivers/SRAM`.

The SPI transport layer is separated from the drivers, and the current implementation uses a register-level "fastspi" implementation.  The "fastspi" implementation imports a "fastspi_config.h" that can be ported to different boards.

There is still some minor future work when additional SRAM drivers are added, but this one has been in the back log forever and is a good foundation.

### [... cut]
closed 2023-12-22 by Jake-Carter