analogdevicesinc/msdk #638: Mac OS Build Failure

Eric-Busch0, 2023-06-24:
I tried to use the MSDK on my mac mini. GCC can't seem to find some file in any of the projects. The Compilation/Debugging portion of the toolchain seems fine as far as I can tell. 

[image]

If you just add "empty.c" to the build directory, compilation is fine.
I don't see it in any of the Makefiles at a project level, so it seems to be either a deeper tier of the Makefile process or a bug with the automatic *.C aggregation.

@Jake-Carter

Jake-Carter, 2023-06-25:
Thanks @Eric-Busch0

The empty file was a recent addition as of #628 (see this commit to gcc.mk https://github.com/Analog-Devices-MSDK/msdk/commit/53527bbcf6a1d622a5e15c83ccef2b4bc9693f50).  We use it to generate a header file that contains build-time and compiler-specific definitions.

It sounds like the `$(file ...)` operation is failing on MacOS.  I'll take a look and replicate.

Can you run a `make VERBOSE=1` into a text file and attach it to this ticket?

Eric-Busch0, 2023-06-25:
[build_output.txt](https://github.com/Analog-Devices-MSDK/msdk/files/11860434/build_output.txt)

Jake-Carter, 2023-06-27:
Found a pretty serious issue that may be affecting the version of Make itself that is running across all our MacOS projects.  (https://github.com/microsoft/vscode/issues/186244)

That's seemingly related, but I couldn't replicate your exact error yet.

Can you follow ["Completing the Installation on MacOS"](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#completing-the-installation-on-macos) from our UG, and then brute-force the environment setup in settings.json?  That was my workaround for the ticket above, but we can't apply it for our user's projects.

It will help determine if an old version of Make is failing to create the empty.c file behind the scenes or if there's something we need to look into with GCC.

Thanks

[image]

Eric-Busch0, 2023-06-27:
Yup. seems to be me not reading the documentation. It is definitely a make issue running gmake as homebrew installs it lets the build work. The pre installed make has some issue. 

gmake --version 
[image]

When using make --version

[image]

Is there a reason for rosetta? It built fine with gmake for both both native and rosetta. 
Last thing about the documentation, Terminal is in Applications/Utilities. Might be confusing for people not used to mac.

Also homebrew like to change their directories like clockwork, but it always says this in the output
[image]

Which gives you the exact path you need to put in the shell startup.

If you need me to test any of this release on my mac let me know.

Jake-Carter, 2023-06-30:
Thanks @EricB-ADI, this should be resolved as of #641 in commit https://github.com/Analog-Devices-MSDK/msdk/pull/641/commits/4505f2440012511c54fd6e6202087b3ab4b89ca1.

The VS Code project files now explicitly source ~/.zshrc on MacOS as a workaround.  It looks like our ticket on the VS Code repo has been added to their backlog, so hopefully this is a temporary fix.

> Is there a reason for rosetta? It built fine with gmake for both both native and rosetta.

Our OpenOCD binaries are the problem

> Last thing about the documentation, Terminal is in Applications/Utilities. Might be confusing for people not used to mac.

I'll update the UG

closed 2023-06-30 by Jake-Carter