analogdevicesinc/msdk #640: 78000FTHR "Insert SD card to continue" Not identifying SD card

ryalberti, 2023-06-27:
I am trying to run both SDHC_weights and facial_recognition on my MAX78000FTHR. I have checked that my Micro SD card is correctly formatted with FAT32 and empty before inserting it into the board with the metal contacts facing the board. When I build the projects, they receive no errors or warnings. When they are loaded, I receive the expected outputs waiting for the SD cards to be inserted. For SDHC, I have tested with two different empty micro SD cards without partitions. I have also tried facial_recognition with empty SD cards as well as with the weights_2.bin file in the root directory and received the same waiting prompt. I have tested this by having the SD card already inserted upon running and inserting the SD card after the program has begun. I have tested this two boards with the same results.

I see [here](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#sdhc) that the MAX780000FTHR is not included in the SDHC usage although I know these programs are both ( [1](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX78000/CNN/facial_recognition#max78000-feather-operations), [2](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX78000/CNN/facial_recognition/SDHC_weights#project-specific-build-notes)) specifically tested for this board. I am developing from the repo and not from the maintenance tool so I am up fully updated as of this morning, 6/27/2023. This leads me to believe there is a setting I have not changed or I am building somehow incorrectly. To build, I am simply running ‘ make -r -j ’ Are there any additional changes to make?  

Thank you!

[image]

[image]

Jake-Carter, 2023-06-28:
Thanks @ryalberti

As part of our functional testing before release I have been tracking any issues in the [fix/functional-tests](https://github.com/Analog-Devices-MSDK/msdk/tree/fix/functional-tests) branch.

Good catch in the UG, I added the MAX78000 to the list of SDHC supported parts (https://github.com/Analog-Devices-MSDK/msdk/pull/641/commits/056b0e51cd29f2f6711a95680df0f1f378a432c4)

On the example - you are building it correctly.  The default build settings will work.

I was able to run the SDHC_Weights project, which formatted and wrote to the card.

```serial
***** MAX78000 Writes CNN weights to SD card *****
Card inserted
SD card mounted.
SD Card Opened
Opened file 'weights_2.bin'
176844 total bytes to write
3000 bytes written to file
3000 bytes written to file
...
3000 bytes written to file
2844 bytes written to file
176844 total bytes written to file
File Closed
SD Unmonted
```

Afterwards, I ran the facial_recognition project successfully.

```serial
D[main      : 103]

MAX78000 Feather Facial Recognition Demo

D[main      : 124] Camera I2C slave address is 3c

D[main      : 134] Camera Product ID is 7692

D[main      : 144] Camera Manufacture ID is 7fa2

Card inserted
SD card mounted.
SD Card Opened
D[facedetection: 158] CNN load data time : 25ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 24ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 25ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 25ms
```

So the example seems to be functional.  Perhaps there is a difference in SD cards that we haven't caught.  Below is the card I'm using.  What is yours?

[image]

Jake-Carter, 2023-06-30:
Thanks @ryalberti, looks like the 32GB capacity is the only major difference.  I'll order some higher capacity cards to test

ryalberti, 2023-06-30:
Hi Jake, I found an 8GB which yielded the same results on my board. 
[image]

ryalberti, 2023-07-05:
I do have a second FTHR board I can test with. The second board used the above Micro Center sd and yielded the same issue, though it used the same build files as the first FTHR that tested the Samsung, SanDisk, and Netac.

Jake-Carter, 2023-07-05:
Can you zip up your .elf file and attach it?  I'll test your binary on my working board

Suraj-Ajjampur, 2023-07-06:
I just flashed a Sandisk Ultra Plus 32 GB micros SDHC card and was unable to reproduce the issue. This is similar to the SDHC card @ryalberti used.
[image]

Suraj-Ajjampur, 2023-07-06:
Try looking at the Updating the MAX32625PICO ("PICO") Debug Adapter Firmware section in this [link](https://github.com/MaximIntegratedAI/MaximAI_Documentation/tree/master/MAX78000_Feather#updating-the-max32625pico-pico-debug-adapter-firmware)

Also, these are the parameters I set before formatting the micro SDHC. 
[image]

ryalberti, 2023-07-06:
I put the listed firmware onto the board and received the correct Git BASH, and copied your parameters when formatting my SD card. I still received the same results. Attached is my .elf from my computer.
[max78000.zip](https://github.com/Analog-Devices-MSDK/msdk/files/11971614/max78000.zip)

On the second board, we build the project from a previous build of the elf (though I have changed no values, I have updated the repo) and one build on another computer with the same update. I will be unable to test the above directions on the other board until tomorrow. 

Thank you for your patience!

ryalberti, 2023-07-07:
I was able to test the second board this morning, flashing with the above settings yielded the same results. Attached is the .elf used for this board. The computer that built this .elf is not developing from the repo, it is using the official release.
[max78000.zip](https://github.com/Analog-Devices-MSDK/msdk/files/11982686/max78000.zip)

Suraj-Ajjampur, 2023-07-11:
I ran the above .elf file and I wasn't able to reproduce that issue. 
I got the below output.

[image]

I think this might still be the way the SDHC is being formatted try running the SDHC_FTHR example (msdk/Examples/max78000/SDHC_FTHR) and select the Format option. Hope this helps.

ryalberti, 2023-07-11:
We tried on both boards and neither of them are detecting an SD card being inserted at all. The first board was testing using the SanDisk (32gb) and the Netac (8gb) and the second was using the same MicroCenter card. I was initially under the impression that if it was two boards, then it must have been a software issue. Does it appear that this is a hardware issue and we should be replacing the board?

[image]

Jake-Carter, 2023-07-11:
Given that flashing your binary on my board also works...

```serial
D[main      : 103]

MAX78000 Feather Facial Recognition Demo

D[main      : 124] Camera I2C slave address is 3c

D[main      : 134] Camera Product ID is 7692

D[main      : 144] Camera Manufacture ID is 7fa2

Card inserted
SD card mounted.
SD Card Opened
D[facedetection: 158] CNN load data time : 25ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 25ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 25ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 25ms
D[facedetection: 170] CNN wait time : 20ms
D[facedetection: 158] CNN load data time : 24ms
D[facedetection: 170] CNN wait time : 19ms
D[facedetection: 158] CNN load data time : 25ms
```

... this has the symptoms of a hardware issue.

1. You've formatted your card with the same options as [above](https://github.com/Analog-Devices-MSDK/msdk/issues/640#issuecomment-1623927089)
2. You've flashed and run the `SDHC_weights` project on the card and that fails
3. You've flashed and run the `SDHC_FTHR` example and that also fails
4. I can run your same binary on my board and it works
5. Multiple SD card sizes work on our side

The symptoms seem to indicate hardware-related failure on the SD card interface.  It's shocking to me that both your boards have it. I've never seen this happen before, but I'm not sure what else to try.

Shoot me an email to jake.carter@analog.com with your shipping info and I'll see about getting you some tested FTHR boards asap.

Jake-Carter, 2023-07-12:
Hardware issue verified

closed 2023-07-12 by Jake-Carter