analogdevicesinc/msdk #654: June 2023 Release Issues

ConWalsh, 2023-07-07:
Hello!

SDK was updated to the latest version (June 23 release) and a number of issues were encountered:

1. No _File -> New -> Maxim Microcontrollers._ in Eclipse anymore. 
[image]

2. New build errors for previously building projects. It looks like some Source/Include folders were removed in Libraries/Boards/MAX_XXXXX that contained high level board code such as pb.c/pb.h. 
[image]

3. If any reference to the above code is removed I am getting the following :
[image]

Thanks

ConWalsh, 2023-07-07:
Following [these steps](https://confluence.analog.com/display/PBUSPKB/Can%27t+See+Maxim+Microcontrollers+Project+in+Eclipse+Maxim+SDK+After+Update) I was able to rectify the first problem.

Jake-Carter, 2023-07-10:
@ConWalsh you didn't miss it, I added it to the release notes thanks to your findings.  So thanks again for your report!

Jake-Carter, 2023-07-10:
We just pushed a hotfix for the Eclipse plugin as well.

As of `1.0.4-31` the new project wizard should be fixed as well and visible in the usual spot.

[image]

[image]

closed 2023-07-10 by Jake-Carter