analogdevicesinc/msdk #671: Makefile example projects don't build with latest Xcode make on macOS Ventura

andrewmclachlanadi, 2023-07-16:
```
make -r -j 8 --output-sync=target --no-print-directory 
/Applications/Xcode.app/Contents/Developer/usr/bin/make: unrecognized option `--output-sync=target'
Usage: make [options] [target] ...
```

Workaround is to tweak the Makefile Eclipse settings to remove `--output-sync=target`

Jake-Carter, 2023-07-17:
See https://analog-devices-msdk.github.io/msdk/USERGUIDE/#completing-the-installation-on-macos

andrewmclachlanadi, 2023-07-17:
Thanks @Jake-Carter .

Is there something specific other than option which requires Make 4.x. If not, then it might be a nicer out-of-the box experience to not have to install the dependent tools or to also include a Make 4.x build in the macOS installation.

Jake-Carter, 2023-07-17:
`output-sync=target` prevents parallel builds from mangling the terminal output, which is a good feature that makes parallel builds actually usable.

I'm in favor of including the Make 4.x binaries in the macOS installation.  We have been limited in our ability to package specific to macOS, but with your help we can address that!

Jake-Carter, 2023-07-18:
Closing as we move towards an installer that packages make for all platforms

yc-adi, 2024-11-30:
The link https://analog-devices-msdk.github.io/msdk/USERGUIDE/#completing-the-installation-on-macos is invalid now. And the problem seems still be there.