analogdevicesinc/msdk #688: Bit position of Presence Detect Flag in OWM_CTRL_STAT register out of place

Suraj-Ajjampur, 2023-07-25:
While integrating the MAX31825 1-Wire temperature sensor with the MAX32666 board, I ran the OWM example in the SDK and noticed that while running the ow_romid_test(), the function returned an error code 2. 

While debugging I was able to isolate the issue to the function 
MXC_OWM_GetPresenceDetect() which is supporsed to return presence pulse detect status. 
 
Although the OWM slave device was driving the signal low (verified by logic analyzer) as shown below.

[image]

This function returns the Presence Detect Flag in the OWM_CTRL_STAT register. 
This is done using the following logic 

```
int MXC_OWM_GetPresenceDetect(void){    
return (!!(MXC_OWM->ctrl_stat & MXC_F_OWM_CTRL_STAT_PRESENCE_DETECT));
}

```
However, the bit position MXC_F_OWM_CTRL_STAT_PRESENCE_DETECT_POS defined in owm_regs.h differs from what is documented in the Userguide of the MAX32666.

[image]

Once I changed the macro definition to the correct bit position the example passed with the example as shown below.

[image]

The OWM is defined for different micros such as ai87, me10, me13, me14, me17 and the me18. After verifying the userguide for all these micros they all have this presence detect flag at bit position 7.
 
Hence changing the MXC_F_OWM_CTRL_STAT_PRESENCE_DETECT_POS in owm_regs.h will be a solution to all these micros.
But we would also have to change this in the .svd files for all the micros.

@sihyung-maxim

sihyung-maxim, 2023-07-25:
Verification testing is required for the AI87, ME10, ME13, ME17, and ME18 as these register fields were updated to 5 within the last year.

Suraj-Ajjampur, 2023-07-26:
Sure, shall run these validation tests for the other micros on Aug 4th.

EricB-ADI, 2024-04-04:
@sihyung-maxim can we close this?