analogdevicesinc/msdk #698: My MAX78000 doesn't show up as a serial port anymore

Samotric, 2023-08-05:
I was originally able to program the board using the Maxim SDK eclipse, but at some point the board stoppped showing in the device manager as a serial port. The board is still programmable but I have no way of seeing the UART results on the serial terminal anymore. I tried unlocking it using the "erase.act" action but that didn't work.

Jake-Carter, 2023-08-07:
Hi @Samotric, which board are you using?  ([MAX78000EVKIT](https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/max78000evkit.html) or [MAX78000FTHR](https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/max78000fthr.html))

Samotric, 2023-08-07:
I am using the EVKIT board

Jake-Carter, 2023-08-07:
Thanks.  Given the board is still programmable it sounds like CN1 is the issue.  Can you confirm?

The COM port that CN1 should present is provided by an [FT230XS-R](https://ftdichip.com/products/ft230xs/) USB to Serial converter (U3 on the [schematic](https://github.com/Analog-Devices-MSDK/msdk/files/12285475/MAX78000EVKIT.Datasheet.pdf)).

Driver issues could be possible.  

A good driver install will show up under Ports (COM & LPT) in Device Manager as a USB Serial _Port_.  (USB Serial _Device_ is the debugger)

[image]

Known working drivers on my machine are

[image]

A bad driver install may cause the port to show up in "Other Devices" category with a yellow warning label on it.  (I don't have any bad devices to test, but here's the category you're looking for)

[image]

Does CN1 show up as anything in "Other Devices" when you plug it in?

Samotric, 2023-08-08:
It is CN1. The board doesnt show up at all in the device manager tho. It used to show as a USB serial port now it doesnt at all.

Jake-Carter, 2023-08-08:
Sounds like possible hardware failure then.  Have you tried a different micro-USB cable yet?

Jake-Carter, 2023-08-14:
@Samotric if you're still seeing failure after trying a different cable I think the symptoms are pointing to a failed FTDI chip.  You could most likely initiate an RMA with your distributor to request a refund/replacement board.

closed 2023-08-14 by Jake-Carter