analogdevicesinc/msdk #699: FreeRTOS BLE "CordioM" task not receiving notifications

khpeterson, 2023-08-06:
While running repeats of my BLE streaming application (based on BLE_FreeRTOS), occasionally I end up in a state where the CordioM task is no longer able to receive notifications.  I traced this to inconsistent state in the FreeRTOS task control block and what seems like a race condition/synchronization bug.  I don't know how to fix the problem but I do know for sure that the CordioM task state variable ucNotifyState is not taskWAITING_NOTIFICATION even though the task is on the xSuspendedTaskList when notification is attempted.  

I instrumented xTaskGenericNotify to check for this condition:
```
            traceTASK_NOTIFY( uxIndexToNotify );

            /* If the task is in the blocked state specifically to wait for a
             * notification then unblock it now. */
            /********************************************************************/
	    /* tasks on delayed list must be waiting for notification */
	    if (listLIST_ITEM_CONTAINER( &( pxTCB->xStateListItem ) ) == &xSuspendedTaskList) {
		//configASSERT(ucOriginalNotifyState == taskWAITING_NOTIFICATION);
		if (ucOriginalNotifyState != taskWAITING_NOTIFICATION) {
			notify_out_of_sync_original_state = ucOriginalNotifyState;
			notify_out_of_sync_count++;
		}
	    }
            /********************************************************************/          
            if( ucOriginalNotifyState == taskWAITING_NOTIFICATION )

```
and then detect and report the condition in my app's run loop:
```
		extern volatile uint32_t notify_out_of_sync_count;
		extern volatile uint8_t notify_out_of_sync_original_state;
		extern uint32_t last_notify_out_of_sync_count;
		if (last_notify_out_of_sync_count != notify_out_of_sync_count) {
			TRACE_ERR2("detected notify out of sync, notify_out_of_sync_count = %d, original_state = %d", notify_out_of_sync_count, notify_out_of_sync_original_state);
			last_notify_out_of_sync_count = notify_out_of_sync_count;
		}
```
While running repeats, sometimes after just a few minutes, sometimes after several hours, I see in my console log:
```
khp@Kevins-MacBook-Pro-M1 % tail -f logs/repeats_20230805_ble_hang.log | grep "out of sync"
2023-08-05T18:18:51.211971: runLoop: detected notify out of sync, notify_out_of_sync_count = 1, original_state = 0
2023-08-05T18:18:56.172692: runLoop: detected notify out of sync, notify_out_of_sync_count = 2, original_state = 2
```
Reviewing the code in Libraries/FreeRTOS/Source/tasks.c it's not clear to me how a task can ever be in the taskNOT_WAITING_NOTIFICATION state and be suspended, but I don't have a lot of experience with this code.  Once in this state the CordioM task will never respond to notifications and BLE is essentially hung.

EdwinFairchild, 2023-08-21:
@khpeterson Hey Kevin, is related to #685 ? Does that PR fix this issues or still persists? Do you have a minimal application you can share so I can try to reproduce?

khpeterson, 2023-08-21:
Hi @EdwinFairchild, thanks for taking a look.  For sure this issue is not related to https://github.com/Analog-Devices-MSDK/msdk/pull/685 - I am running with my fork which includes that pull request.

Unfortunately, I do not have a pared down app that will easily reproduce the dropped notifications.  From what I can tell the issue is timing dependent and given that it sometimes takes hours of continuous BLE streaming with my full app to see, it may be pretty hard to create one.  

I have a lot on my plate at the moment but my first thought would be to pull out the Wsf message queue/dispatch and send/receive messages continuously with 2 tasks with varying load conditions (e.g. random durations via MXC_Delay).

My suspicion is that the critical section protection set up in the port of FreeRTOS is not bullet proof, but I can't prove that.

My current work around is to force both the msg task and the event task back into the waiting notification state when I detect the out of sync condition.  With this (and https://github.com/Analog-Devices-MSDK/msdk/pull/685) in place I can stream BLE over night without a problem.

khpeterson, 2023-08-27:
Hi @EdwinFairchild, I was able to reproduce this with a minimal application (attached).  Please build with [this branch of my fork](https://github.com/khpeterson/analog-devices-msdk-fork/tree/assert-notify-out-of-sync) to pick up the change for the wsfTimer leak as well as an assertion that checks for (ucOriginalNotifyState == taskWAITING_NOTIFICATION) when the task being notified is on the suspended list.

I'm running on a vanilla MAX32666FTHR with a shell logging from the UART console connecting to BLE using ADI Attach (version 1.0.1+2 on an iOS 16.6 iPhone 10) : connect to DATS and enable notifications for the "ARM Proprietary Data Service" and let it run.  Timing varies, and it may take a couple of hours, but eventually you will see the assertion I added to FreeRTOS/Source/tasks.c.

[BLE_FreeRTOS_v2023_06_notif_out_of_sync.zip](https://github.com/Analog-Devices-MSDK/msdk/files/12447277/BLE_FreeRTOS_v2023_06_notif_out_of_sync.zip)

khpeterson, 2025-05-23:
Hi @EdwinFairchild, it has been a while and I know this is not a high priority for you, but FWIW, this problem still persists with the latest SDK (f98b65b120fff8f46f057e70125e6499f454cdcd, as of 5/14/2025).  I sync'd up my fork of the msdk tree and overlaid the changes to the BLE_FreeRTOS app to stress task notifications and committed them, along with the assertion described above (in FreeRTOS/tasks.c, ucOriginalNotifyState == taskWAITING_NOTIFICATION when a task being notified is on the suspended list), to https://github.com/khpeterson/analog-devices-msdk-fork/tree/assert-notify-out-of-sync.

To summarize how to reproduce:

- clone https://github.com/khpeterson/analog-devices-msdk-fork/tree/assert-notify-out-of-sync
- build BLE_FreeRTOS, BOARD=FTHR, TARGET=MAX32665 from this branch
- flash to a vanilla MAX32666FTHR
- log the debug console UART
- connect to the "DATS" BLE device with a logging tool
- enable notifications on the E0262760-08C2-11E1-9073-0E8AC72E1001 characteristic
- let it stream, likely for several hours and eventually you will see something like the following in the debug console:
```
.
.
.
2025-05-22T23:58:00.075588: Dats got evt 154, event_count = 786764
2025-05-22T23:58:00.083011: Dats got evt 154, event_count = 786765
2025-05-22T23:58:00.097418: Dats got evt 154, event_count = 786766
2025-05-22T23:58:00.101092: Dats got evt 154, event_count = 786767
2025-05-22T23:58:00.105974: Dats got evt 18, event_count = 786768
2025-05-22T23:58:00.109926: Dats got evt 154, event_count = 786769
2025-05-22T23:58:00.114408: Dats got evt 154, event_count = 786770
2025-05-22T23:58:00.118392: Dats got evt 154, event_count = 786771
2025-05-22T23:58:00.123488: Dats got evt 154, event_count = 786772
2025-05-22T23:58:00.128805: Dats got evt 153, event_count = 786773
2025-05-22T23:58:00.135955: Dats got evt 154, event_count = 786774
2025-05-22T23:58:00.141959: Source/tasks.c #4923: (ucOriginalNotifyState == taskWAITING_NOTIFICATION)
```

EdwinFairchild, 2025-05-24:
Hey @khpeterson ,
Sorry to hear the issue is still present in the latest SDK. I do not recall where I left off with this. However I now work for STMicroelectronics. This should be on their radar if they are still tracking github issues.

khpeterson, 2025-06-02:
A little more color on this issue - here's a more detailed trace around the time of the unexpected resume with task name filtering disabled (was previously limiting the notify trace macros to "CordioM" as the source/target):

|  Pal RTC Counter | Task Name | Trace Point | Comment |
| - | - | - | - |
| 575356162 |    MY_TASK | traceMOVED_TASK_TO_READY_STATE | xTaskIncrementTick |
| 575356162 |    MY_TASK | traceTASK_SWITCHED_IN          |  |
| 575356162 |       IDLE | traceTASK_SWITCHED_IN          |  |
| 575356194 |    Tmr Svc | traceMOVED_TASK_TO_READY_STATE | xTaskIncrementTick |
| 575356194 |    MY_TASK | traceMOVED_TASK_TO_READY_STATE | xTaskIncrementTick |
| 575356195 |    Tmr Svc | traceTASK_SWITCHED_IN          |  |
| 575356195 |    Tmr Svc | traceTASK_NOTIFY               |  |
| 575356195 |    CordioM | traceMOVED_TASK_TO_READY_STATE | xTaskGenericNotify |
| 575356195 |    CordioM | traceTASK_SWITCHED_IN          |  |
| 575356195 |    CordioM | traceTASK_NOTIFY_WAIT          |  |
| 575356197 |    CordioM | traceTASK_NOTIFY_WAIT_BLOCK    |  *** NOT yielding |
| 575356198 |    CordioM | traceTASK_NOTIFY_WAIT          | *** falling through, yielding |
| 575356198 |    Tmr Svc | traceTASK_SWITCHED_IN          |  |
| 575356199 |    MY_TASK | traceTASK_SWITCHED_IN          |  |
| 575356199 |       IDLE | traceTASK_SWITCHED_IN          |  |

I have been pouring over the code and traces and from my view the only thing that explains this trace and the fact that "CordioM" task is still on the suspended list is that it is not blocking in the top half of xTaskGenericNotifyWait() where it should yield and force a context switch through PendSV_Handler at the exit from the first critical section.  Instead it is falling through to the bottom half of the function and then yielding on exit from the second critical section. 

I still can't explain why this is happening but will keep digging.

bstafford549, 2025-06-28:
It looks like you found the root cause of the FreeRTOS & Cordio hang and created a fix. Excellent!