analogdevicesinc/msdk #710: BLE_periph example fails to connect

Vitorbnc, 2023-08-15:
The provided BLE_periph example does not connect on Windows 10 and Android, and the device is not visible on iOS.

The BLE_fit example seems to work OK. The iPhone pop-up says it didn't connect but shows the device status as Connected later on and the phone and BLE peripheral keep exchanging packets.

Apparently there is no encryption on the periph example and an error occurs when calculating the database hash. Could we add some minimal encryption to the periph example so it would connect?

[image]

I have attached detailed logs with the connection attempts.
[logs.zip](https://github.com/Analog-Devices-MSDK/msdk/files/12346932/logs.zip)

Vitorbnc, 2023-08-25:
I think maybe we could clarify it in the readme, that the BLE_periph example isn't meant to work with Windows/Android/iOS, but only with other embedded devices.

I have implemented a full UART over BLE terminal based on the BLE_mcs and the BLE_fit example over this week.
There are some other subtleties that could be made more explicit in the documentation. For instance, besides the need for security features as you said, iOS ignored any custom 128-bit UUID, it would only show the device when advertising with a 16-bit Bluetooth SIG UUID.

EricB-ADI, 2023-09-22:
@Vitorbnc We will update the README as soon as we can. In the meantime, feel free to update them and create a pull request.

closed 2023-10-24 by EricB-ADI