analogdevicesinc/msdk #731: Argument List Too Long - RISC-V AR

BrentK-ADI, 2023-09-15:
When building the Cordio library for the RISC-V on MAX32690, the AR command fails with an "Argument List too Long" error.  It looks like this was resolved in gcc.mk by using an ar_args.txt for the arguments.  This should be ported over to the RISC-V makefile as well.

Jake-Carter, 2023-09-21:
Thanks @BrentK-ADI, looks like I fixed this issue when I synced up this GCC makefiles with each other in https://github.com/Analog-Devices-MSDK/msdk/pull/653

The release SDK doesn't have these changes yet, the development repo should fix this issue.  I'll close this out when we push the release in case anyone else hits it.

Jake-Carter, 2023-10-11:
Closing alongside the Oct 2023 release:  https://github.com/Analog-Devices-MSDK/msdk/releases/tag/v2023_10

closed 2023-10-11 by Jake-Carter