analogdevicesinc/msdk #736: MAX32675 IRQ vector table in max32675.h: entries for interrupts that do not exist

oledreessen, 2023-09-19:
in the IRQ vector table for the MAX32675 are entries for peripherals that do not exits. 
examples:
RTC_IRQn,    /* 0x13  0x004C  19: RTC */
TMR5_IRQn,   /* 0x1A  0x0068  26: Timer 5 */
UART1_IRQn,  /* 0x1F  0x007C  31: UART 1 */
SPI2_IRQn,   /* 0x22  0x0088  34: SPI2 */     
I2C1_IRQn,   /* 0x34  0x00D0  52: I2C1 */

The above list is not complete.

EricB-ADI, 2023-09-22:
Hi @oledreessen , many of our chips that have different part numbers use the same die. The MAX32675 is a mix of a MAX32670 and a die that has no part number as it is not a standalone block. The IRQs that you have mentioned lie within the reserved regions mentioned in the datasheet as they are used in the MAX32670. In order to get the added functionality of the newer die, some blocks are left unconnected to free up space for the the new features. Therefore they still exist, but will not do anything. We generate our register files based on the die not the part number. So they still exist in the headers.

Jake-Carter, 2023-09-29:
Thanks @oledreessen and @EricB-ADI.  Will close this out for now since it doesn't affect functionality.

@sihyung-maxim FYI - perhaps this is something we can address with the register file generator in the future.

closed 2023-09-29 by Jake-Carter