analogdevicesinc/msdk #761: UART Leaks DMA Channels, Fails After 8 DMA Transactions

Jake-Carter, 2023-10-06:
Our Master DMA transaction functions internally acquire a DMA channel, but never release it or expose it to the application code.  After 8 transactions, all DMA channels have been consumed and subsequent transactions will fail to transmit any data.

This is a common design issue we've seen recently in a few other peripherals, and we should apply similar fixes.

aniktash, 2023-10-09:
Also, MXC_DMA_ReleaseChannel(ch_num) only accepts the DMA channel, not the DMA instance 0 or 1 as in MAX32666

Jake-Carter, 2023-10-09:
Fixed the channel leak with the following diff:

```diff
diff --git a/Libraries/PeriphDrivers/Source/UART/uart_reva.c b/Libraries/PeriphDrivers/SouhDrivers/Source/UART/uart_reva.c
index 10f85e0374..2bf5469a40 100644
--- a/Libraries/PeriphDrivers/Source/UART/uart_reva.c
+++ b/Libraries/PeriphDrivers/Source/UART/uart_reva.c
@@ -996,21 +996,15 @@ void MXC_UART_RevA_DMACallback(int ch, int error)        
     mxc_uart_reva_req_t *temp_req;

     for (int i = 0; i < MXC_UART_INSTANCES; i++) {
-        if (states[i].channelTx == ch) {
-            //save the request
-            temp_req = states[i].req;
-            // Callback if not NULL
-            if (temp_req->callback != NULL) {
-                temp_req->callback((mxc_uart_req_t *)temp_req, E_NO_ERROR);   
-            }
-            break;
-        } else if (states[i].channelRx == ch) {
+        if (states[i].channelTx == ch || states[i].channelRx == ch) {        
             //save the request
             temp_req = states[i].req;
             // Callback if not NULL
             if (temp_req->callback != NULL) {
                 temp_req->callback((mxc_uart_req_t *)temp_req, E_NO_ERROR);   
             }
+
+            MXC_DMA_ReleaseChannel(ch);
             break;
         }
     }
```

Jake-Carter, 2023-10-09:
Working on getting the transactions working across all DMA channels now.

Jake-Carter, 2023-10-10:
It's a miracle that the old drivers work with DMA at all...  They are completely reliant on the right handler being defined by chance. 

I have a solution implemented on [fix/gh-761](https://github.com/Jake-Carter/msdk/tree/fix/gh-761) where the drivers create the DMA handlers automatically for the user.

I tested DMA0 and DMA1 with the following code on the MAX32666FTHR.

```C
/******************************************************************************
 * Copyright (C) 2023 Maxim Integrated Products, Inc., All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a
 * copy of this software and associated documentation files (the "Software"),
 * to deal in the Software without restriction, including without limitation
 * the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the
 * Software is furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included
 * in all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS
 * OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
 * MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.
 * IN NO EVENT SHALL MAXIM INTEGRATED BE LIABLE FOR ANY CLAIM, DAMAGES
 * OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE,
 * ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 * OTHER DEALINGS IN THE SOFTWARE.
 *
 * Except as contained in this notice, the name of Maxim Integrated
 * Products, Inc. shall not be used except as stated in the Maxim Integrated
 * Products, Inc. Branding Policy.
 *
 * The mere transfer of this software does not imply any licenses
 * of trade secrets, proprietary technology, copyrights, patents,
 * trademarks, maskwork rights, or any other form of intellectual
 * property whatsoever. Maxim Integrated Products, Inc. retains all
 * ownership rights.
 *
 ******************************************************************************/

/**
 * @file    main.c
 * @brief
 * @details 
 * /

/***** Includes *****/
#include <stdio.h>
#include <stdint.h>
#include <string.h>
#include "mxc_device.h"
#include "pb.h"
#include "board.h"
#include "uart.h"
#include "led.h"

/***** Definitions *****/
#define DMA_INSTANCE MXC_DMA0
#define BUFF_SIZE 512
#define AUTO_HANDLERS

/***** Globals *****/
volatile int UART_DONE;
volatile int buttonPressed;

/***** Functions *****/

/******************************************************************************/
void UART_Callback()
{
    UART_DONE = true;
}

void buttonHandler()
{
    buttonPressed = 1;
}

/******************************************************************************/
int main(void)
{
    int error, i;
    uint8_t TxData[BUFF_SIZE];

    LED_On(0);
    printf("\n\n**************** UART DMA Stream Test ******************\n");
    printf("Push SW1 to continue\n\n");

    buttonPressed = 0;
    PB_RegisterCallback(0, (pb_callback)buttonHandler);
    while (!buttonPressed) {}

    printf("UART Baud \t: %d Hz\n", CONSOLE_UART);
    printf("Test Length \t: %d bytes\n\n", BUFF_SIZE);

    // Initialize the TX data buffer
    for (i = 0; i < BUFF_SIZE; i++) {
        TxData[i] = i;
    }

    // Initialize the UART
    error = MXC_UART_Init(MXC_UART_GET_UART(CONSOLE_UART), CONSOLE_BAUD, TERMINAL_UART_MAP);
    if (error < E_NO_ERROR) {
        printf("-->Error initializing UART: %d\n", error);
        printf("-->Example Failed\n");
        return error;
    }

    printf("-->UART Initialized\n\n");

    while(1) {
        printf("\nPush SW1 to transmit a stream...\n");
        buttonPressed = 0;
        while (!buttonPressed) {}

        mxc_uart_req_t write_req;
        write_req.uart = MXC_UART_GET_UART(CONSOLE_UART);
        write_req.txData = TxData;
        write_req.txLen = BUFF_SIZE;
        write_req.rxLen = 0;
        write_req.callback = UART_Callback; // Note callback function is now working/preferred way to poll for TX finish

        UART_DONE = false;
        error = MXC_UART_TransactionDMA(&write_req, DMA_INSTANCE);

        if (error != E_NO_ERROR) {
            printf("-->Error during transaction: %d\n", error);
            printf("-->Example Failed\n");
            return error;
        }

        while (!UART_DONE) {}
    }    
}

```

The current solution isn't totally compatible with existing (mostly broken) code, so I'm working on an API for users who want to define their own handlers and channels.

PR #763 (2023-11-30): fix(PeriphDrivers): Fix UART DMA Transactions and Enable Full Duplex
### Description

Our UART DMA transactions were essentially unusable, and only happened to work if the transaction function happened to acquire DMA channel 0.

This PR fixes #761 

It implements 2 features:
1. UART DMA "Auto handlers"
2. APIs for using application-defined DMA handlers and channels

### DMA Auto Handlers

**This behavior is disabled by default...** but **I would _like_ to make this the default**.  Please comment.

Users now have the option to enable "auto handlers" for the UART DMA transactions with

```C
MXC_ [... cut]