analogdevicesinc/msdk #767: I2S Example for the MAX32666

selimgullulu, 2023-10-12:
We don't have an I2S example for the MAX32665/6 in the MSDK. Can this be implemented?
Regards
Selim

Jake-Carter, 2023-10-18:
Hi @selimgullulu, actually the MAX32665 family does not have an I2S peripheral.  I2S is supported as a sub-mode of a custom "audio subsystem" (see section 23 of the [MAX32665 UG](https://github.com/Analog-Devices-MSDK/msdk/pull/726)).

I checked our SDK and it looks like we don't even have register-level support for this audio subsystem.  We define the base address, but none of the other registers.  I will check with the team to see what the status of that peripheral is.

selimgullulu, 2023-10-20:
Hi @Jake-Carter thank you for your update.
Regards
Selim

Jake-Carter, 2023-10-25:
Hi @selimgullulu,

Apparently we lost contact with the original design team for this peripheral some time ago...  but the peripheral is not listed as defeatured in any of our public facing documentation, so it's reasonable that we have some level of support for it in the MSDK.

It will take some time to track down the design resources for it, but @sihyung-maxim has started to look into it.

ersahatmacaa, 2023-10-26:
Hi @Jake-Carter 

I just started working in the Customer Applications team in Turkey. Right now, I'm writing a low-level example project for the audio subsystem. But I'm only using I2S transmit and receive mode on PCM. If it's alright, I would also like to contribute.

Regards
Ersah

Jake-Carter, 2023-10-28:
Hi @ersahatmacaa, welcome aboard!  Glad to have you.

We would love your contributions.  Do you have the register-level documentation you need?

ersahatmacaa, 2023-10-30:
Hi @Jake-Carter, thank you. 

I have the user guide, which is public. Apart from that, if there is any documentation related to the audio subsystem, I would like to check it.

selimgullulu, 2024-01-30:
I think the issue can be closed since @ersahatmacaa has introduced the Audio_Playback example with https://github.com/Analog-Devices-MSDK/msdk/pull/815

closed 2024-02-07 by Jake-Carter