analogdevicesinc/msdk #772: Problems about linking a new lib to a proj

liuyuxivapor, 2023-10-17:
I got trapped in linking a new lib, CMSIS-DSP for example, to an existing project. I have set the include path where CMSIS-DSP/include locates but an error occurred on building. 
`main.c:7:10: fatal error: arm_math.h: No such file or directory`
Another question is about how to use hardware DSP directly, I find it hard to look for repos or examples about this.
Thanks a lot if anyone could help me!

Jake-Carter, 2023-10-17:
Hi @liuyuxivapor,

For CMSIS-DSP, see the [`ARM-DSP`](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX78000/ARM-DSP) examples folder for each micro.  We have a prebuilt CMSIS-DSP library you can use if you'd like.

For a general example about how to link an external library, see the [`Library_Use`](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX78000/Library_Use) example.  In general, the following needs to be added to your _project.mk_ file:

```Makefile
# project.mk

# Add include path to header files
IPATH += lib/include
# Add search folder to locate the .a file
PROJ_LDFLAGS += -L/some/library/path
# Add library file to PROJ_LIBS variable.
# Ex:  libmyLib.a
# Ignore the "lib" and ".a" in the filename
PROJ_LIBS += myLib
```

Most of the configuration you need is for the build system.  More documentation can be found in the [MSDK User Guide](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#build-system).  For example, to set hardware floating point acceleration you can set [`MFLOAT_ABI = hard`](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#build-variables-for-arm-cortex-m4-cores)

liuyuxivapor, 2023-10-18:
Got it. Thanks a lot.

closed 2023-10-18 by liuyuxivapor