analogdevicesinc/msdk #774: ImgCapture Example with MAX78000FTHR - No Image received

ABx911, 2023-10-17:
I am using the MAX78000FTHR and tried to run the ImgCapture example with the SDK. 

I was able to get the example up, running on the MAX without any problems and receive some images. One week later nothing works like before. In Putty I got the following output:

[image]

I then started the ImgCapture console on Python (3.9.7). It seems to work correctly. But after I tried to run the registred commands nothing happened. I didn't neither receive an image or informations nor any error. The board is unable to receive any commands.

[image]

I´ve tried the example on different computers and attempted a deinstallation of the SDK and a Board-Reset. It still did not work as desired.

Is this an issue related to my building of the program or my SDK install? What can I do to receive images again?
Thank you in advance!

Jake-Carter, 2023-10-17:
Hi @ABx911, it looks like the firmware is getting stuck in the camera initialization, most likely in [this](https://github.com/Analog-Devices-MSDK/msdk/blob/d6eb11cacef202f054624eab22f107264bc4bf00/Libraries/MiscDrivers/Camera/camera.c#L467) while loop.  

I just tested with the latest SDK again to confirm things are still working:

[image]

I can replicate the same behavior if I attempt to run firmware built for the MAX78000EVKIT instead of the MAX78000FTHR.

So first thing to check is to make sure you've set the Board Support Package correctly for the MAX7800FTHR, then clean and rebuild the project.  See the [MSDK User Guide](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#board-support-packages) for instructions.

FYI this example initializes the serial port at 921600 to improve the data transfer rate for the images.  A baudrate mismatch would explain the mangled output.  It also prints out "*SYNC*" until the host terminal echoes it back.

ABx911, 2023-10-17:
Hi @Jake-Carter, thank you for your fast reply!

I've changend the Board Support Package from EvKit to FTHR and also change the data transfer rate.
After doing that I've got the right output at Putty with the "SYNC" you've mentioned. 

However, I still cannot receive any images. I noticed that in the Python console already the I2C slave address of the camera and the selected camera ID are not output. Therefore I don't get the message that the camera is ready. I think that is also why I can not receive images. What can be the cause for this?

ABx911, 2023-10-17:
I've tried the I2C-Scanner example to see if any I2C device and especially the camera could be found. But the scanner found no device. So as a logical conclusion I can't get any images either if the camera is not recognized as an I2C device.

What can I do here? Is something on the board broken?

Jake-Carter, 2023-10-18:
For this example you should use the python `console.py` script instead of Putty.  The Python script has some image conversion code to save the streamed data.

But that wouldn't be the root cause of the issue you're having...

Just to confirm:
- You set your BSP to `FTHR_RevA`
- You cleaned and rebuilt your project
- The Python script connects but fails to initialize the camera

Can you paste the output of the Python script?

It's possible that the camera module itself has problems, but I would expect any communication issues to throw an error on the camera initialization.  The first thing the script will do is read out the camera ID over I2C and verify it.  Any issues with the I2C will cause that verification to fail.

You can also attach your binary (.elf file) and I can run it on my board to rule out any build issues.

ABx911, 2023-10-24:
@Jake-Carter could you find out something?

Jake-Carter, 2023-10-24:
Hi @ABx911 thanks for the ping - I did test your binary and it seemed to work fine on my board.

[image]

I suspect this may be some ESD and/or hardware issues with your camera module.  I have seen a bad module every now and then, and I've also seen ESD cause I2C communication issues in the past.

Since your binary works on my board I think we can at least rule out software/build issues

Jake-Carter, 2023-10-25:
@ABx911 I'll close this out as probable ESD/hardware failure.  Hopefully you can still RMA the board with your distributor and get a new one

Let me know if not and I can see if I can submit a sample order request for you, but I'll probably need to get some business/application info from you to justify it to the orders team.  My email's jake.carter@analog.com

Best of luck in your development!

closed 2023-10-25 by Jake-Carter