analogdevicesinc/msdk #784: Update request for SDHC default SPI Speed

oguzhanbsolak, 2023-10-26:
Hi,

For the upcoming MAX78000FTHR facial recognition demo, we need a 25MHz SPI for SDHC to get a higher FPS.
Currently, this value is set to 10MHz at the SDHC/ff13/Source/diskio.c by the SPI_SPEED parameter. Is it possible to update the default value to 25MHz or add a new flag to enable high-speed spi usage for SDHC?

Regards,
Oguzhan Buyuksolak

Jake-Carter, 2023-10-27:
We can also support this.  I'll add the ability to override the default SPI speed per-project.

oguzhanbsolak, 2023-10-30:
Hi Jake,

Thank you for your return on these issues.

Regards,
Oğuzhan Büyüksolak

closed 2023-11-06 by Jake-Carter

Jake-Carter, 2023-11-09:
@oguzhanbsolak FYI this is merged.  See the PR description above for details on the new build variable

PR #791 (2023-11-06): feat(SDHC): Add Ability to Override SDHC_CLK_FREQ
### Description

Closes #784 

The SDHC clock frequency can now be overridden with the new `SDHC_CLK_FREQ` build configuration variable.  Since the SDHC library is a larger library with multiple components, this warrants a build variable instead of a simple compiler definition.  

The default is 40Mhz.  Changing the value will affect the native and SPI-emulated SDHC implementations.  Ex:

```Makefile
# project.mk

SDHC_CLK_FREQ  = 25000000
```

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description  [... cut]