analogdevicesinc/msdk #786: MSDK Maintenance Tool RISC-V/Visual Studio Code Support Co-Dependency

jlj-ee, 2023-10-27:
Minor headache here: 
At present, when I set up the MSDK installation, I exclude the RISC-V Embedded GCC component because I do not need it for my current projects/targets, and it requires a lot more disk space than the other components.
However, if I try to also install the Visual Studio Code Support component  (which I do want/need), it seems to implicitly select the RISC-V component to download & install. Likewise, on a machine where I had installed RISC-V, uninstalling RISC-V apparently uninstalled VS Code support. Is this expected behavior?
I am on Windows.

Steps to reproduce:
1. Run the maintenance tool on a system where the MSDK is already installed.
2. De-select RISC-V and select Visual Studio Code support
[image]
3. Go to next, note that the installer displays the license agreement for RISC-V
[image]
4. Accept, note that the installer displays that it will be installing a component the same size as the RISC-V component (4.49GB as of the October 2023 MSDK release).
[image]

I worked around this by just grabbing the VS Code support release from the VSCode-Maxim repo, but it seems like a bug in the Maintenance Tool/Installer.

Jake-Carter, 2023-10-27:
Hi @jlj-ee, the VS Code component has the RISC-V toolchain as a dependency because it relies on it to compile the RISC-V projects.

Unfortunately the QT installer framework enforces dependencies in both directions, so removing the RISC-V component will also remove any components that depend on it, including MAX78000/MAX78002 part support...

The behavior is expected, but in this case it's a pretty big limitation.  Considering the disc space I think we can consider removing the direct dependencies for the toolchain components so users can uninstall them if needed, and just leave them as enabled by default.

Jake-Carter, 2023-10-27:
Speaking of the VSCode-Maxim repo - this reminded me I needed to tag the latest v1.6.2 release.

Just created a release page for it:  https://github.com/Analog-Devices-MSDK/VSCode-Maxim/releases/tag/v1.6.2

Jake-Carter, 2024-02-12:
Staged in [v1.7.0 pre-release](https://github.com/Analog-Devices-MSDK/VSCode-Maxim/releases/tag/v1.7.0-pre), target release date for the next MSDK update is Feb 16th

closed 2024-03-19 by Jake-Carter