analogdevicesinc/msdk #798: Cordio AppDb eventually runs out of space and then becomes unusable

sullivanmj, 2023-11-16:
When dealing with bonded bluetooth devices, the AppDb that is part of Cordio can run out of space. This happens if AppDb records are written enough times, which can happen under typical use for a long enough time. Once this happens, a `WSF_ASSERT` will be hit during the next attempted read or write. This causes the core to halt. A `WSF_ASSERT` will also be hit during subsequent (post-reboot) calls to `WsfNvmInit()`, which will prevent the Bluetooth application from initializing.

Jake-Carter, 2023-11-21:
@EdwinFairchild can you take a look?

EdwinFairchild, 2023-11-28:
To clarify , are you establishing bonded connections with a lot of devices and running out of NVM space? If so then I agree we should handle running out of NVM  gracefully.

sullivanmj, 2023-11-28:
Yes, this can happen as a result of just repeatedly creating new bonds.

You can use all NVM space faster by it by reducing `PAL_NVM_SIZE`, or by increasing `APP_DB_NUM_CCCD` to increase the size of each record that gets stored in the DB.

sullivanmj, 2024-04-01:
Can you please provide an ETA on a fix for this issue? We would like to avoid going into production with this bug in place.

Jake-Carter, 2024-04-01:
Hi @sullivanmj, Eddie has transitioned out of ADI

I will reach out internally to re-assign this ASAP to get you an updated ETA

Jake-Carter, 2024-04-03:
We've opened a preliminary fix in #975, where instead of an assertion error we will return an error code and log a trace message.

@EricB-ADI and @yc-adi have started to pick up where Eddie left off addressing the core memory issue.

@sullivanmj - do you have a timeline when you're targeting a move into production?

EricB-ADI, 2024-04-03:
@Jake-Carter I have a PR open right now where the code will return an error if the flash fills up. I am currently working on trying to make some code to de-fragment the flash.

sullivanmj, 2024-04-03:
Thanks, all. I will be happy to help test the MR when the time comes.

EricB-ADI, 2024-04-04:
@sullivanmj  The code is ready on #975. I have tested it and it all works, but feel free to test it in your use case. I am pulling it in to main, Friday 4/5, if you do not get a chance to take a peak.

sullivanmj, 2024-04-04:
Thanks. I tested out the code that checks for space when writing to the NVM. I left a couple comments on the MR. I'm working my way through testing the defrag code now. I found a tengentially related issue while looking into that. `PalNvmGetTotalSize` returns the size in ints rather than the size in bytes, because of its pointer math and the fact that those symbols are defined as 32-bit integers. Not sure if we should file another issue for that or whether it could be fixed in your PR as well.

EricB-ADI, 2024-04-05:
That is a very subtle and important bug. I am tacking it into this PR.

EricB-ADI, 2024-04-10:
Fixed in #975

closed 2024-04-10 by EricB-ADI