analogdevicesinc/msdk #801: Issues with ternary operator & operator precedence in Cordio

sullivanmj, 2023-11-20:
I was running a static code analyzer on my project and it produced warnings that there are a few places in Cordio where there may be some bugs due to the ternary operator `?:` having a lower precedence than the addition operator `+`.

In each of these cases, the ternary operator should looks like it should be evaluated before the addition operator, but it is not.

https://github.com/Analog-Devices-MSDK/msdk/blob/4461e6d6383ac7e31285534e4e4e1a87531e5baf/Libraries/Cordio/controller/sources/ble/lctr/lctr_main_conn.c#L1568

https://github.com/Analog-Devices-MSDK/msdk/blob/4461e6d6383ac7e31285534e4e4e1a87531e5baf/Libraries/Cordio/controller/sources/ble/lctr/lctr_main_adv_master_ae.c#L1307

https://github.com/Analog-Devices-MSDK/msdk/blob/4461e6d6383ac7e31285534e4e4e1a87531e5baf/Libraries/Cordio/controller/sources/ble/lctr/lctr_int_adv_master_ae.h#L477

Jake-Carter, 2023-11-21:
Thanks @sullivanmj, I've just opened a PR to add some extra parenthesis.

Let us know if your analyzer uncovers anything else

sullivanmj, 2023-11-21:
No problem - that was the only thing it uncovered.

PR #803 (2023-11-28): fix(BLE): Add Parenthesis Around Some Cordio Ternaries
## Pull Request Template

### Description

Closes #801

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [ ] (Optional) Link any related GitHub issues [using a keyword](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword)
- [ ] (Optional) Provide info on any relevant functional testing/validation.  For API changes or significant features, this [... cut]
closed 2023-11-28 by Jake-Carter