analogdevicesinc/msdk #821: Some questions about "NUM CLASSES"

PYL4869, 2023-12-08:
hi
the SVHN TinierSSD model used for the “digit-detection-demo” project
In svhn.py, 
[image]

In post_process.h,
[image]

Why is "NUM CLASSES" equal to 12, shouldn't it be 10 ( 1, 2, 3, 4, 5, 6, 7, 8, 9, 10 (0) )+1 (background) = 11?
Does it serve any special purpose? Or I got it wrong

PYL4869, 2023-12-10:
A few more questions:
1：How are these addresses calculated? Why do we execute "cl_addr += 0x003e0000;" at a specific address? Go to a specific address? How should "0x50823000" "0x50c23000" "0x003e0000" be calculated?
 
[image]

2：In "ai85-facedet-tinierssd.yaml" and "ai87-facedet-tinierssd.yaml", why can both "CHW" and "HWC" be used? Does that mean it doesn't matter what the inputs are when the model is trained?

[image]
[image]

ermanok, 2023-12-12:
> hi the SVHN TinierSSD model used for the “digit-detection-demo” project In svhn.py, ![Snipaste_2023-12-09_00-39-27](https://private-user-images.githubusercontent.com/103161086/289134064-62e8371c-2a1b-4e13-a873-bbb6a5fad736.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0MTgzNDAsIm5iZiI6MTcwMjQxODA0MCwicGF0aCI6Ii8xMDMxNjEwODYvMjg5MTM0MDY0LTYyZTgzNzFjLTJhMWItNGUxMy1hODczLWJiYjZhNWZhZDczNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxMlQyMTU0MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZjFiNDY4NmU2ZDZmMTg1MWQyYjkzOGZhMWJkMGI2ZThjNDQ0ODc0OWM0MWRjNzM2YTg5ZDJkYWZkNjYzNzEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.64le6lgIxkpn6BEkvp4Bc6tvxys_h0QeoOo36QDjzFQ)
> 
> In post_process.h, ![image](https://private-user-images.githubusercontent.com/103161086/289134997-3f979c07-7b7e-47bc-a969-58197b1fc90d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0MTgzNDAsIm5iZiI6MTcwMjQxODA0MCwicGF0aCI6Ii8xMDMxNjEwODYvMjg5MTM0OTk3LTNmOTc5YzA3LTdiN2UtNDdiYy1hOTY5LTU4MTk3YjFmYzkwZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxMlQyMTU0MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NzRhY2NkYjljOWE3MWY4ZjJjMWJiMWNkZWVlZTY3NDFkZGNmMDg0OTZkZWIzY2RmMzU2NGIwOGFlNTZhNzBjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.LAmVUnxQHlWXRv_Tjst78RsGXnGCC0rX6psy-W8FiTs)
> 
> Why is "NUM CLASSES" equal to 12, shouldn't it be 10 ( 1, 2, 3, 4, 5, 6, 7, 8, 9, 10 (0) )+1 (background) = 11? Does it serve any special purpose? Or I got it wrong

As you mentioned, the parameter "NUM_CLASSES" is expected to be 11 and the implementation can be done so. It is set to 12 on purpose to unload the class predictions of the model easier from the data memory. As explained in the documentation (https://github.com/MaximIntegratedAI/ai8x-training?tab=readme-ov-file#data-memory), each data memory block stores 4 channels of the data. So it is easier to fetch the data in 4 channels blocks from the memory. In other words, the model has 11 channels output as you expect but the class predictions of the model is fetched as if it has 12 channels outputs just for easier implementation. If you check, the Non Maxima Suppression (NMS) functionality, it ignores the last (12th) channel.

ermanok, 2023-12-12:
For your next questions here are my answers:

1: This is about the map of the data memory on the chip (see https://www.analog.com/media/en/technical-documentation/user-guides/max78000-user-guide.pdf pp. 382). The output data is written starting from addr 0x50803000 and the first 4 channels of the data is in this data memory block. Once the data is read, the pointer is set to addr 0x50823000. So it should be moved to the next data memory block, which is at 0x50c03000. This is why, the data pointer is moved 0x3e0000 element further. After the data in this block is read, the data pointer is at 0x50c230000, it should be moved to the next data memory block at 0x51003000.

2: The field "data_format" defines how you write the input data to the data memory. Therefore, it is independent of how you train the model. In both HWC and CHW modes, the model consumes the same input data. But you have to set the processors of the layer accordingly (as in the examples you shared). Please see the README (https://github.com/MaximIntegratedAI/ai8x-training?tab=readme-ov-file#channel-data-formats) for the details.

PYL4869, 2023-12-13:
> For your next questions here are my answers:
>
> 1: This is about the map of the data memory on the chip (see https://www.analog.com/media/en/technical-documentation/user-guides/max78000-user-guide.pdf [...]

hi,Thank you for your reply, If my "classes_scores" is like this, and 12 similar addresses are not generated in "cnn_unload", how should I modify it? I tried to modify it, but the effect is not good, maybe I don't understand.
[image]
[image]
[image]

ermanok, 2023-12-14:
As I understand, you have two output layers with 24 channels each and different spatial resolutions 12x15 & 3x3. The data addresses you will read these output data is determined in the yaml file you used to synthesize the code. The procesor map and the 'output_processors' are the fields you need. If the 'output_processors' is 0xffffff0000000000, and the 'out_offset' is '0x3510', the first 4 channels data is at the addresses from 0x50C13510 to 0x50C137E0 (12x15x4 = 720 bytes). The next 4 channels data is at 0x50C1B510 to 0x50C1B7E0. The next 4 channels data is at the next memory block so it is at 0x51003510 to 0x510037E0 however the pointer cl_addr is at (0x50C13510 + 0x8000 + 0x8000) = 0x50C23510., which is why you have to add 0x51003510-0x50C23510 = 0x003E0000 not 0x003E8000. 

So your code should change as:
	if ((cl_addr == (int8_t *)0x50C23510) ) {
                cl_addr += 0x003e0000;
            }

By the way, you have to set the 'out_offset' of the layer with 3x3 output to '0x37E0' and keep the 'output_processor' field same as the previous layer to make the running code correct. It assumes all the memory address of the output layers are set back to back.

PYL4869, 2023-12-15:
> As I understand, you have two output layers with 24 channels each and different spatial resolutions 12x15 & 3x3. The data addresses you will read these output data is determined in the yaml file you [...]

hi
I followed your advice and modified it accordingly, It works, but it sucks. I modified "post_process.c" and "post_process.h" according to the model. I tried to change "MAX_PRIORS", "MIN_CLASS_SCORE" and "MAX_ALLOWED_OVERLAP", but it still didn't work. The mAP can reach 0.934 and the yaml file is fine.
[image]
[image]

I think "yaml", "post_process.c" and "post_process.h" may need to be modified. Could you help me to check what else needs to be modified? Attached are the relevant documents
[about.zip](https://github.com/Analog-Devices-MSDK/msdk/files/13687180/about.zip)

PYL4869, 2023-12-20:
> Did you compile and run the KAT project (the code generated by the izer) on MAX78000 board? If so, could you please share me your checkpoint file (.pth.tar) and the bash command you use to run the izer? Let me check the code on my side thoroughly... Thanks.

hi, I added some new ones, "app tft.c" may vary depending on the device, You can focus on "void App Cnn Execute(void)", It was too big, so I uploaded it to Google Cloud Drive. I think the reason for the poor effect may be due to the insufficient number of data sets I made by myself.
[https://drive.google.com/file/d/19THJ6Wr1LmJLGJfcKfPkAciIzz1IghY4/view?usp=sharing](url)

ermanok, 2023-12-28:
Hi again,

I found two issues with the files you shared with me and they are related to the estimation of box locations:
1.  In the yaml file, the out_offset of Layer 13 (loc_conv12_2) is set to 0x4000. However, in the function 'get_prior_locs' the location estimations of the layers 12 & 13 are fetched like they are located back to back at the memory. If the out_offset of Layer 13 is set to 0x0x37E0, the problem is solved.
2. The SVHN demo app is developed for square input. So the dimensions of the boxes are estimated assuming the layers 12&13 inputs are square. In your case, the input is 100x120 so you need to modify the box location estimation code accordingly. Rather than defining just "const int dims[NUM_SCALES] = {12, 3};", you need to define "const int dims_y[NUM_SCALES] = { 12, 3};
const int dims_x[NUM_SCALES] = { 15, 3 };". Then you have to modify the code accordingly. Our FPN detector uses non square input and its NMS code has a similar change. You can copy the related changes from there (https://github.com/Analog-Devices-MSDK/msdk/blob/main/Examples/MAX78002/CNN/pascalvoc-retinanetv7_3/src/cnn/nms.c).

PYL4869, 2024-01-10:
> Hi again,
>
> I found two issues with the files you shared with me and they are related to the estimation of box locations:
>
> 1. In the yaml file, the out_offset of Layer 13 (loc_conv12_2) is set to [...]

Hello, thank you for your help, I think I should be able to finish it.

closed 2024-01-10 by PYL4869