analogdevicesinc/msdk #823: MAX32670 MXC_TMR_Init CTRL1_CLKSEL_A/B mismatch

jlj-ee, 2023-12-08:
At least for the MAX32670, MXC_TMR_Init does not seem to properly configure the peripheral in 32 bit mode if the clock source is anything other than the default MXC_TMR_APB_CLK (MXC_TMR_CLK0).
Specifically, CTRL1_CLKSEL_B doe not get set to match CTRL1_CLKSEL_A, so if any clock source other than the default is used (e.g. IBRO/MXC_TMR_CLK2), CLKSEL_B will be different than A and 32 bit mode will not work properly unless the field is set manually after calling MXC_TMR_Init.

see e.g. [MXC_TMR_RevB_Init](https://github.com/Analog-Devices-MSDK/msdk/blob/547540b2bb09a3e7c8f7572ef2c73d6f7c9888f9/Libraries/PeriphDrivers/Source/TMR/tmr_revb.c#L100)

Jake-Carter, 2023-12-12:
Thanks for reporting @jlj-ee, just opened a PR with the fix.  If you wouldn't mind testing it I'd greatly appreciate it!

jlj-ee, 2023-12-15:
Thanks for the heads up @Jake-Carter! The fix works well, at least for the particular scenario where it was initially encountered (IBRO clock source for PWM in 32b mode).

PR #829 (2023-12-22): fix(PeriphDrivers): Fix TMR Rev B Initialization Bug with Non-Default Clock Sources
### Description

Closes #823 

There's a new note about clock selection in Rev 4 of the MAX32670 UG (See section 13.4)

[image]

In the past, the settings for the "A" timer were assumed to propagate through to the "B" timer in 32-bit cascade mode. It seems like there was a bug, so now we should match the clock selection for A/B on initialization to ensure proper operation in the 32-bit modes.

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all [... cut]