analogdevicesinc/msdk #824: I2C Example Enhancement for MAX32690EVKIT

selimgullulu, 2023-12-11:
The I2C example fails to run by the default jumper settings on the MAX32690EVKIT. The reason for this is the Jumper 9 and Jumper10 are set to CTS and RTS by default on the EVKIT. These jumpers have to be set to SDA and SCL. 
The explanation in the main() can be extended with this information:
 printf("\n******** I2C Master-Slave Transaction Demo *********\n");
    printf("\nThis example uses one I2C peripheral as a master to\n");
    printf("read and write to another I2C which acts as a slave.\n");

    printf("\nYou will need to connect P1.8->P2.8 (SCL) and\n");
    printf("P1.7->P2.7 (SDA).\n");
    **printf("Make sure the Jumper 9 and Jumper 10 on the MAX32690EVKIT are set to SDA and SCL");_**
[image]

Jake-Carter, 2023-12-12:
Thanks @selimgullulu, the [README](https://github.com/analogdevicesinc/MAX32xxx-MSDK/blob/main/Examples/MAX32690/I2C/README.md#required-connections) does highlight setting JP9/JP10, but I've just updated the printfs to make this more clear as well.

selimgullulu, 2023-12-13:
Hi @Jake-Carter , thanks for the update!

PR #828 (2023-12-22): fix(Examples): Highlight JP9/JP10 in printfs for MAX32690EVKIT I2C Example
### Description

Closes #824 

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [ ] (Optional) Link any related GitHub issues [using a keyword](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword)
- [ ] (Optional) Provide info on any relevant functional testing/validation.  For API changes or significant features, this is n [... cut]
closed 2023-12-22 by Jake-Carter