analogdevicesinc/msdk #826: can we compile MAX78000 projects with G++ instead of GCC?

khavernathy, 2023-12-12:
I'd like to use C++ code for a project. Changing `main.c` to `main.cpp` in one of the sample projects results in a bunch of `undefined reference to ...` errors.

Jake-Carter, 2023-12-12:
Thanks for reporting this @khavernathy, I've just opened a PR to fix this.  We were missing C++ guards in some of our display and camera files.

C++ is fully supported, and the camera drivers should compile with the PR changes.

khavernathy, 2023-12-12:
Thanks, for what it's worth, these are all the undefined references I get when changing `main.c` to `main.cpp` and attempting to build:

```
/home/x/dev/x/max78000/main.cpp:164: undefined reference to `camera_start_capture_image()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:168: undefined reference to `camera_get_image(unsigned char**, unsigned long*, unsigned long*, unsigned long*)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:209: undefined reference to `release_camera_stream_buffer()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:176: undefined reference to `get_camera_stream_buffer()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:177: undefined reference to `camera_is_image_rcv()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:216: undefined reference to `get_camera_stream_statistic()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/build/main.o: in function `transmit_capture_uart()':
/home/x/dev/x/max78000/main.cpp:297: undefined reference to `clear_serial_buffer()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:304: undefined reference to `send_msg(char const*)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:310: undefined reference to `clear_serial_buffer()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:313: undefined reference to `MXC_UART_WriteBytes(mxc_uart_regs_t*, unsigned char const*, int)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/build/main.o: in function `main':
/home/x/dev/x/max78000/main.cpp:358: undefined reference to `cnn_enable(unsigned long, unsigned long)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:361: undefined reference to `cnn_boost_enable(mxc_gpio_regs_t*, unsigned long)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:364: undefined reference to `cnn_init()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:366: undefined reference to `cnn_load_weights()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:368: undefined reference to `cnn_load_bias()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:370: undefined reference to `cnn_configure()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:378: undefined reference to `camera_init(unsigned long)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:380: undefined reference to `camera_setup(int, int, pixformat_t, fifomode_t, dmamode_t, int)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:390: undefined reference to `camera_write_reg(unsigned char, unsigned char)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:422: undefined reference to `cnn_unload(unsigned long*)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:423: undefined reference to `cnn_stop()'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:426: undefined reference to `softmax_q17p14_q15(long const*, unsigned short, short*)'
/home/x/MaximSDK/Tools/GNUTools/10.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/x/dev/x/max78000/main.cpp:413: undefined reference to `cnn_start()'

```

PR #827 (2023-12-22): fix(MiscDrivers): Add Missing C++ Guards to Camera and Display Files
## Pull Request Template

### Description

Closes #826 

### Checklist Before Requesting Review

- [ ] PR Title follows correct guidelines.
- [ ] Description of changes and all other relevant information.
- [ ] (Optional) Link any related GitHub issues [using a keyword](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword)
- [ ] (Optional) Provide info on any relevant functional testing/validation.  For API changes or significant [... cut]
closed 2023-12-22 by Jake-Carter