analogdevicesinc/msdk #858: Instructions for flashing MAX78000 over SPI

khavernathy, 2024-01-08:
Are there any docs for flashing the chip through SPI? (instead of through a host PC using `arm-none-eabi-gdb` and `openocd`)

For example could I make an executable using the `*.elf` that is made through the build process and then write it to the chip at address `0x0`?

Jake-Carter, 2024-01-08:
In general this is called a "bootloader", and the MAX78000 does have a built-in bootloader documented in section 26 of the [User Guide](https://github.com/Analog-Devices-MSDK/msdk/issues/709).  It exposes a UART interface.  I would recommend this option if you'd like to do in-application programming, though the documentation does have some outstanding issues.

See these tickets for more details and resources:
- https://github.com/Analog-Devices-MSDK/msdk/issues/485
- https://github.com/Analog-Devices-MSDK/msdk/issues/709

It's also possible to write your own bootloader and put it in a reserved section of flash, but this is a more advanced option.

Is SPI a strict requirement for the bootloader interface?

khavernathy, 2024-01-10:
SPI doesn't seem to be a strict requirement. 

ty for helpful info. I'll do some research to see if I can figure out flashing over UART.

closed 2024-02-08 by Jake-Carter