analogdevicesinc/msdk #865: Error: Target not examined yet

ethyljoybadiang, 2024-01-11:
I got this using both Eclipse and VSCode.  I am trying to deploy Hello_world on MAX78002EVKIT. I have actually tried it already, so as faceid, kws20_demo, etc. but suddenly I got this error in both Eclipse and VSCode.
[image]
[image]

Jake-Carter, 2024-01-13:
Hi @ethyljoybadiang, considering you've been able to run some demos already this is most likely a symptom of low power lockout.  See the documentation on [Debug Limitations](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#debug-limitations) for more details.  There is a recovery procedure you can attempt to recover the part.

Let me know if it works for you.

Jake-Carter, 2024-01-13:
FYI the default port used to connect to OpenOCD from Eclipse is 3333.  Same for VS Code's `flash` and `flash & run` tasks.  The Cortex-Debug profile uses 50000 by default.

However, the connections are made through the localhost.  So it's very rare (but not impossible) for the ports to be blocked.

ethyljoybadiang, 2024-01-13:
> Hi @ethyljoybadiang, considering you've been able to run some demos already this is most likely a symptom of low power lockout. See the documentation on [Debug [...]

Okay, will try that now. I also tried running the examples from MAX78000 using my MAX78000FTHR board running it in Eclipse using a different laptop connected to a different network, same error showed.

ethyljoybadiang, 2024-01-13:
I have tried it already (erasing the flash bank). It does not contain, fail.txt, but it is also not blank as it should be (as stated in 10).

ethyljoybadiang, 2024-01-13:
So, this is the comparison after I did it.
[image]

Jake-Carter, 2024-01-15:
The DETAILS.TXT file looks good.

In VS Code, can you paste the output of the "erase flash" task here?

Additionally, after running the "flash" or "flash & run" tasks there should be a `flash.log` file in your project.  Can you attach it here as well?

This will give me some more detailed info to see what's going on.

ethyljoybadiang, 2024-01-16:
Here is the flash.log file.
[flash.log](https://github.com/Analog-Devices-MSDK/msdk/files/13943935/flash.log)

ethyljoybadiang, 2024-01-16:
Here is the output after the erase flash.
[image]

ethyljoybadiang, 2024-01-16:
I tried building, flashing, and debugging it again after erase flash. I got this. 
[image]

Jake-Carter, 2024-01-16:
Thanks @ethyljoybadiang.  OpenOCD is at least connecting to the micro in some capacity.  Usually any low-power firmware lockout will throw the same error message as if your SWD cable is disconnected, or the board is powered off.  Usually that's a "Cannot read IDR" message.

In this case, the primary error message is "Failed to read memory and, additionally, failed to find out where".  This is usually a more esoteric problem and more difficult to debug.  I wonder if you have been aware of the power-sequencing requirements for the MAX7002EVKIT?  (See the quick-start guide [here](https://github.com/MaximIntegratedAI/MaximAI_Documentation/tree/master/MAX78002_Evaluation_Kit#power-sequencing)) It's possible that the chip could be damaged in some capacity, especially since you were able to run some demos on it before.

Do you remember the last demo you ran?

ethyljoybadiang, 2024-01-16:
I once had wrong power sequencing but that was before I had trouble. Also, same errors show in the featherboard.

ethyljoybadiang, 2024-01-16:
Can it be the ESD?

ethyljoybadiang, 2024-01-16:
[image]

Jake-Carter, 2024-01-16:
The PICO debugger can back-power into the board, which can exceed the absolute max ratings on some components (including the MAX78002) if the right power-up/power-down sequencing isn't followed.  It looks like you've got the PICO debugger connected to the SWD port, right?  And the Olimex connected to the RV JTAG port?  (see [Connecting Debug Adapters](https://github.com/MaximIntegratedAI/MaximAI_Documentation/tree/master/MAX78002_Evaluation_Kit#connecting-debug-adapters) in the QSG).

It looks like the [faceid_evkit demo](https://github.com/Analog-Devices-MSDK/msdk/tree/main/Examples/MAX78002/CNN/faceid_evkit) is flashed.  Is the demo actually running?  (i.e. it's refreshing the screen with data from the camera?)

If the demo is running another thing you can try is holding the "RESET" button, then spamming the "erase flash" task in VS Code _immediately_ after releasing reset.  The debugger will not be able to connect to the micro _while_ reset is held, but there is a [2 second delay](https://github.com/Analog-Devices-MSDK/msdk/blob/261425f8012dcb2b2e396360713f04a7c5fee828/Examples/MAX78002/CNN/faceid_evkit/main.c#L82) placed at the beginning of the code to provide the debugger a window to connect to.  If you hit this window (and the part is not damaged by the PICO back-powering), you should see a successful erase.

Additionally, can you run the following command in your VS Code terminal and paste the result?

```shell
openocd --version
```

It should match

```shell
Open On-Chip Debugger 0.11.0+dev-g56a818e4c (2023-10-09-16:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```

Finally, please run the following command from your VS Code project.  It will create an `openocd_debug.log` file with even more debug info.  Please attach it to the ticket as well.

```shell
openocd -d3 -s "%MAXIM_PATH%/Tools/OpenOCD/scripts" -f "interface/cmsis-dap.cfg" -f "target/max78002.cfg" -c "init; reset halt" > openocd_debug.log 2>&1
```

Jake-Carter, 2024-01-23:
Hi @ethyljoybadiang, any luck timing the hardware reset?

ethyljoybadiang, 2024-01-23:
Hello! I am still trying to exhaust any software possibilities because the error goes both EVkit and fthrboard. I was told it might be a synthesis error, so I am trying to delve into that. Is it okay if I send my synthesized code if it works on yours?

Jake-Carter, 2024-01-23:
Yes, please zip up your project and attach it to this ticket and I will test it on my board

ethyljoybadiang, 2024-01-23:
Thank you.
[banana.zip](https://github.com/Analog-Devices-MSDK/msdk/files/14018020/banana.zip)

ethyljoybadiang, 2024-01-24:
> The PICO debugger can back-power into the board, which can exceed the absolute max ratings on some components (including the MAX78002) if the right power-up/power-down sequencing isn't followed. It [...]

I actually did not do this, and have not really touched the kit since the last time. I just deleted the workspace and built another one.. 
[image]

Can you explain why? So, the chip is not destroyed?

ethyljoybadiang, 2024-01-24:
Also, it only works on few of the examples in the sdk (Hello_world, faceid, imagenet)

Jake-Carter, 2024-01-31:
Hi @ethyljoybadiang,

The banana known answer test you provided above works fine on my board as well.

```serial
Waiting...

*** CNN Inference Test banana ***

*** PASS ***

Approximate inference time: 3895 us

Classification results:
[ 199214] -> Class 0: 100.0%
[-244409] -> Class 1: 0.0%
```

If you send me one of the projects that's not working for you I will test it as well.

ethyljoybadiang, 2024-02-01:
Ahhh yes thank you. It worked on my board as well after I deleted the eclipse-workspace. Thank you. This case is closed now.

closed 2024-02-01 by Jake-Carter