analogdevicesinc/msdk #872: flash.mk openocd script can't open target binary

Brandon-Hurst, 2024-01-13:
Attempting to flash my MAX78000FTHR via "make flash.openocd" as described in the [User Guide](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#getting-started-with-command-line-development).  
[image]

Here is the output I consistently get:
[image]

I'm able to manually run openocd/arm-none-eabi-gdb server and flash the binary just fine, but the flashing script referenced here is unable to open the binary file. I also tried this from an Administrator-level bash environment.

ethyljoybadiang, 2024-01-14:
Hello, I also have trouble similar to that. May I just ask, did you also encounter a gdb connection error?

Brandon-Hurst, 2024-01-14:
Hi Ethyl,

No, I did not. I was able to flash my app over gdb-openocd. I ran it like it explains in the User Guide, below for your reference:

In one terminal window, run:
`openocd -s ${MAXIM_PATH}/Tools/OpenOCD/Scripts -f interface/cmsis-dap.cfg -f target/max78000.cfg`

In another terminal, cd into your project directory and run `arm-none-eabi-gdb` then run the following commands:
`target extended-remote :3333` connects to your running openocd client
`file build/max78000.elf` will set the target binary to load
`load` will load the flash

I was also able to flash with just OpenOCD by essentially running the same command as the flash.mk but on the command line. 

Incidentally, I also installed onto a VM for Ubuntu 22.04 yesterday and the flash.mk process works just fine on Linux. So it's perhaps a Windows issue.

Jake-Carter, 2024-01-15:
Thanks @Brandon-Hurst, just replicated the issue.  It does seem to be specific to Windows/MSYS2

closed 2024-01-15 by Jake-Carter

Brandon-Hurst, 2024-01-15:
Yep, tested and working. Thanks Jake!

PR #876 (2024-01-15): fix(Build): Fix broken `make flash.openocd` on Windows + MSYS2
### Description

The MSDK includes [Makefile targets for flashing](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#flashing-on-the-command-line) via a `flash.mk` file.  This target depends on a rule for creating a .hex file.  The .hex file is added as an explicit dependency of the flashing targets so that it's built before attempting the flash.  On Windows + MSYS2 the file path was mal-formed.  `cygpath` is required to correct the path, but the code that detected the availability of `cygpath` was not working correctly.

This  [... cut]