analogdevicesinc/msdk #883: MAX78000FTHR Board Can't Flash

nyasar-code, 2024-01-18:
Hi, 

I have a MAX78000FTHR board with a Adafruit 2.4" 320x240 TFT display attached to it. I was running facial_recognition example on it and everything was working well. But something happened and I can't flash my board anymore, it is giving the error:

Remote communication error.  Target disconnected.: Connection reset by peer.

If I erase the flash it gives the error:

Error: unable to find a matching CMSIS-DAP device

After searching through other user's issues, I realized others had the similar error and the solution was to "unlock the board" to make it programmable again. I followed all the steps and everything went smoothly. I updated the debugger and put the erase.act file into the drive and there was no fail.txt generated.

But my issue still persists, I still get the same error as before. Could there be something additional I have to do because of the TFT display? What else can cause this issue? Thanks in advance.

Copy of my Details.txt file:

# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0444170199e18a7600000000000000000000000097969906
HIC ID: 97969906
Auto Reset: 1
Automation allowed: 1
Overflow detection: 1
Page erasing: 1
Daplink Mode: Interface
Interface Version: 0256
Git SHA: a209a92c65513989f86ac7ff32c1bec27082eee0
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0xb61afbb5
Remount count: 0
URL: http://www.maximintegrated.com/max78000fthr

nyasar-code, 2024-01-28:
Hi Jake, 

I have seen you suggest this to other users experiencing the same issue but unfortunately it hasnt worked for me.

Using the board was a part of a college project. I have returned the board back to university and they have given me a new one that is working without any issues. Thanks for your help nonetheless.

Jake-Carter, 2024-02-06:
Hi @nyasar-code, since we have had enough similar issues I spent some time digging into this further.

For anyone else having this issue - 

I found that there is a bug in the default factory DAPLINK firmware (Git SHA a209a92c65513989f86ac7ff32c1bec27082eee0).  That version of the firmware **does not drive the RSTN pin at all**...  It attempts to reset only with the software command.  Therefore it will be unable to reset the device at all from any of the low-power modes.

The latest version (Git SHA 34fe95d79afeba62ef097774ecc609ef8c2b74ea ([DETAILS.TXT](https://github.com/Analog-Devices-MSDK/msdk/files/14173342/DETAILS.TXT))) _does_ drive the RSTN pin correctly and should be able to recover from the lockouts ([download here](https://github.com/analogdevicesinc/max32625pico-firmware-images/raw/master/bin/max32625_max78000fthr_if_crc_swd_v1.0.3.bin)).  I replicated a lockout and where SHA a209 failed, SHA 34fe succeeded.

See the [Debuggers](https://analog-devices-msdk.github.io/msdk/USERGUIDE/#debuggers) section of the UG for update instructions.  The relevant pushbutton to put the MAX78000FTHR in MAINTENANCE mode is SW3.

Jake-Carter, 2024-02-06:
Re-opening for visibility.  I will submit a request to the factory to update the default PICO firmware.

Jake-Carter, 2024-02-13:
The default firmware has been updated to 34fe95 and our existing stock being updated as well.  Closing out, thanks for reporting.

closed 2024-02-13 by Jake-Carter

yuval5295, 2024-04-26:
Hey,

I have the same problem but I am using Max32660EVSYS.
I am using VS (Eclipse made me too much troubles) and unfortunately I had never succeed to flash it.
I did all the instructions step by step - updated the firmware adapter firmware and erase but still no success. 

this is my details on DAPALINK:
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 04211702bfb8038600000000000000000000000097969906
HIC ID: 97969906
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Incompatible image detection: 0
Page erasing: 1
Daplink Mode: Interface
Interface Version: 0256
Git SHA: 34fe95d79afeba62ef097774ecc609ef8c2b74ea
Local Mods: 1
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0xd7617efe
Remount count: 0
URL: http://www.maximintegrated.com/max32660

this is what written on terminal:

[image]
he terminal: