analogdevicesinc/msdk #894: MAX78000 example on MAX78002

ethyljoybadiang, 2024-01-27:
Would it be okay to use demos from MAX78000 in MAX78002? My goal is to do a similar demo to catsdogs but the board I have is MAX78002EVKIT.

Jake-Carter, 2024-01-31:
Hi @ethyljoybadiang, most of the CNNs designed for the MAX78000 can also be run on the MAX78002, but the demos are not directly compatible.  You'll need to re-synthesize the model using the `--device MAX78002` flag (see [command line arguments](https://github.com/MaximIntegratedAI/ai8x-synthesis?tab=readme-ov-file#command-line-arguments-3)).  

The MAX78002 has different hardware and a different register layout, so the model must be re-synthesized to account for that.  Otherwise, the MAX78000-specific `cnn.c` will attempt to write into registers that don't match.

The pre-trained checkpoint files for the MAX78000 can be used to re-synthesize a model for the MAX78002, so no re-training should be required for the pre-made models.  However, if you are training a custom model, you should use `--device MAX78002` with `train.py` as well.

Now, the _demo_ projects that actually interface with camera/TFT displays, etc. will most likely not be directly transferable to the MAX78002.  Some modifications are required to account for differences in the evaluation platforms.

Resynthesizing the cats-dogs known answer test to target the MAX78002 should be as simple as running the following command from the ai8x-synthesis repo.

```shell
 python ai8xize.py --device MAX78002 --test-dir ./synthesized --prefix cats-dogs --checkpoint-file trained/ai85-catsdogs-qat8-q.pth.tar --config-file networks/cats-dogs-hwc-no-fifo.yaml --softmax --timer 0 --display-checkpoint --verbose
```

```shell
~/repos/ai8x-synthesis (develop*) » python ai8xize.py --device MAX78002 --test-dir ./synthesized --prefix cats-dogs --checkpoint-file trained/ai85-catsdogs-qat8-q.pth.tar --config-file networks/cats-dogs-hwc-no-fifo.yaml --softmax --timer 0 --display-checkpoint --verbose 
Configuring device: MAX78002
Reading networks/cats-dogs-hwc-no-fifo.yaml to configure network...
WARNING: Cannot run "yamllint" linter to check networks/cats-dogs-hwc-no-fifo.yaml
Reading trained/ai85-catsdogs-qat8-q.pth.tar to configure network weights...
Checkpoint for epoch 157, model ai85cdnet - weight and bias data:
 InCh OutCh  Weights         Quant Shift  Min  Max    Size Key                                       Bias       Quant  Min  Max Size Key
    3    16  (48, 3, 3)          8    -1  -96  127     432 module.conv1.op.weight                    N/A            0    0    0    0 N/A                      
   16    32  (512, 3, 3)         8    -1  -98   91    4608 module.conv2.op.weight                    N/A            0    0    0    0 N/A                      
   32    64  (2048, 3, 3)        8    -1 -120   93   18432 module.conv3.op.weight                    N/A            0    0    0    0 N/A                      
   64    32  (2048, 3, 3)        8    -1 -104  103   18432 module.conv4.op.weight                    N/A            0    0    0    0 N/A                      
   32    32  (1024, 3, 3)        8     0  -64   76    9216 module.conv5.op.weight                    N/A            0    0    0    0 N/A                      
   32    16  (512, 3, 3)         8     0  -69   61    4608 module.conv6.op.weight                    N/A            0    0    0    0 N/A                      
 1024     2  (1, 2, 1024)        8    -2  -93  101    2048 module.fc.op.weight                       (2,)           8  -14   10    2 module.fc.op.bias        
TOTAL: 7 parameter layers, 57,778 parameters, 57,778 bytes
Configuring data set: cats_vs_dogs.
cats-dogs...
Arranging weights... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
Storing weights...   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
Creating network...  ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
```

Here is the synthesized project from my machine in case you need a known working reference:  [cats-dogs-max78002.zip](https://github.com/Analog-Devices-MSDK/msdk/files/14117942/cats-dogs-max78002.zip)

  Running on my MAX78002EVKIT...
```serial
Waiting...

*** CNN Inference Test cats-dogs ***

*** PASS ***

Approximate inference time: 3896 us

Classification results:
[ -26608] -> Class 0: 11.1%
[  30558] -> Class 1: 88.9%

```

ethyljoybadiang, 2024-02-01:
> Hi @ethyljoybadiang, most of the CNNs designed for the MAX78000 can also be run on the MAX78002, but the demos are not directly compatible. You'll need to re-synthesize the model using the `--device MAX78002` flag (see [command line arguments](https://github.com/MaximIntegratedAI/ai8x-synthesis?tab=readme-ov-file#command-line-arguments-3)).
> 
> The MAX78002 has different hardware and a different register layout, so the model must be re-synthesized to account for that. Otherwise, the MAX78000-specific `cnn.c` will attempt to write into registers that don't match.
> 
> The pre-trained checkpoint files for the MAX78000 can be used to re-synthesize a model for the MAX78002, so no re-training should be required for the pre-made models. However, if you are training a custom model, you should use `--device MAX78002` with `train.py` as well.
> 
> Now, the _demo_ projects that actually interface with camera/TFT displays, etc. will most likely not be directly transferable to the MAX78002. Some modifications are required to account for differences in the evaluation platforms.
> 
> Resynthesizing the cats-dogs known answer test to target the MAX78002 should be as simple as running the following command from the ai8x-synthesis repo.
> 
> ```shell
>  python ai8xize.py --device MAX78002 --test-dir ./synthesized --prefix cats-dogs --checkpoint-file trained/ai85-catsdogs-qat8-q.pth.tar --config-file networks/cats-dogs-hwc-no-fifo.yaml --softmax --timer 0 --display-checkpoint --verbose
> ```
> 
> ```shell
> ~/repos/ai8x-synthesis (develop*) » python ai8xize.py --device MAX78002 --test-dir ./synthesized --prefix cats-dogs --checkpoint-file trained/ai85-catsdogs-qat8-q.pth.tar --config-file networks/cats-dogs-hwc-no-fifo.yaml --softmax --timer 0 --display-checkpoint --verbose 
> Configuring device: MAX78002
> Reading networks/cats-dogs-hwc-no-fifo.yaml to configure network...
> WARNING: Cannot run "yamllint" linter to check networks/cats-dogs-hwc-no-fifo.yaml
> Reading trained/ai85-catsdogs-qat8-q.pth.tar to configure network weights...
> Checkpoint for epoch 157, model ai85cdnet - weight and bias data:
>  InCh OutCh  Weights         Quant Shift  Min  Max    Size Key                                       Bias       Quant  Min  Max Size Key
>     3    16  (48, 3, 3)          8    -1  -96  127     432 module.conv1.op.weight                    N/A            0    0    0    0 N/A                      
>    16    32  (512, 3, 3)         8    -1  -98   91    4608 module.conv2.op.weight                    N/A            0    0    0    0 N/A                      
>    32    64  (2048, 3, 3)        8    -1 -120   93   18432 module.conv3.op.weight                    N/A            0    0    0    0 N/A                      
>    64    32  (2048, 3, 3)        8    -1 -104  103   18432 module.conv4.op.weight                    N/A            0    0    0    0 N/A                      
>    32    32  (1024, 3, 3)        8     0  -64   76    9216 module.conv5.op.weight                    N/A            0    0    0    0 N/A                      
>    32    16  (512, 3, 3)         8     0  -69   61    4608 module.conv6.op.weight                    N/A            0    0    0    0 N/A                      
>  1024     2  (1, 2, 1024)        8    -2  -93  101    2048 module.fc.op.weight                       (2,)           8  -14   10    2 module.fc.op.bias        
> TOTAL: 7 parameter layers, 57,778 parameters, 57,778 bytes
> Configuring data set: cats_vs_dogs.
> cats-dogs...
> Arranging weights... ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
> Storing weights...   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
> Creating network...  ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100%
> ```
> 
> Here is the synthesized project from my machine in case you need a known working reference: [cats-dogs-max78002.zip](https://github.com/Analog-Devices-MSDK/msdk/files/14117942/cats-dogs-max78002.zip)
> 
> Running on my MAX78002EVKIT...
> 
> ```
> Waiting...
> 
> *** CNN Inference Test cats-dogs ***
> 
> *** PASS ***
> 
> Approximate inference time: 3896 us
> 
> Classification results:
> [ -26608] -> Class 0: 11.1%
> [  30558] -> Class 1: 88.9%
> ```

Thank you. Well then, I'll have to delve in to the "Now, the demo projects that actually interface with camera/TFT displays, etc. will most likely not be directly transferable to the MAX78002. Some modifications are required to account for differences in the evaluation platforms." Thank you.

Jake-Carter, 2024-02-10:
Thanks @ethyljoybadiang, will close this out.  Let me know if you have any other questions.

closed 2024-02-10 by Jake-Carter