analogdevicesinc/msdk #902: Bootloader and Bootloader Host Examples for MAX32665 are under Bluetooth Folder

selimgullulu, 2024-02-02:
The Bootloader:
Examples/MAX32665/Bluetooth/Bootloader
and the Bootloader Host:
Examples/MAX32665/Bluetooth/Bootloader_Host
Examples are under Bluetooth folder.
Is there a specific reason for this or is it related to a misclassification (the reason that Bootloader having a "BL" prefix maybe)?
I believe it's better to locate these examples in the root folder of the 32655 Part as the following:
Examples/MAX32665/Bootloader
Examples/MAX32665/Bootloader_Host

sihyung-maxim, 2024-02-05:
The Bootloader examples were originally added to help with the Bluetooth OTA development. Though now, it's considered separate so they can be moved out.

sihyung-maxim, 2024-02-05:
My mistake. The Bootloader example is not considered separate. There are some dependencies with the BLE OTA and Bluetooth examples.

Investigating as to why that's still the case and if we can make them independent of each other.

EdwinFairchild, 2024-02-06:
The Bootloader example is OTAS dependent and specific, its not mean to be a generic bootloader its simply to showcase OTAS BLE example.

closed 2024-02-08 by Jake-Carter

selimgullulu, 2024-02-09:
Thank you all for your explanations.